arduino-libraries/ArduinoBLE #172: Nano 33 BLE in bootmode after disconnect

cansimkp, 2021-04-10:
Hi,
I have successfully paired an AcgamR1 clone remote control with the nano 33 ble.
But after 45 seconds the remote control disconnects. Probably to save energy.
Now the nano hangs up and does nothing more.
No serial monitor and the Builtin LED is blinking 5 quick, 3 slow, 5 quick, 3 slow...
I think the nano is in bootmode. (https://forum.arduino.cc/index.php?topic=630834.0)

Code is from here: https://github.com/bitbank2/Nano_33_Gamepad

Does anyone have a solution for this?

Software:
Arduino IDE 1.8.13
<ArduinoBLE.h> 1.2.0
Arduino mbed -enabled Boards V2.0.0 and same issue with 1.3.2
Hardware:
Nano 33 BLE Original not a clone
BLE Remote: https://www.amazon.de/magicsee-R1-Bluetooth-Wireless-Controller-Joystick/dp/B01MDM48W9

paulvha, 2021-04-11:
I had this problem as well. The root cause is in HCI.cpp, void HCIClass::handleEventPkt(). When EVT_DISCONN_COMPLETE, it will try to HCI.leSetAdvertiseEnable(0x01); while you never have set anything to advertise. ArduinoBLE seems to assume you are running as a peripheral. 

**Solution**
Around line 574 in HCI.cpp, change 
`HCI.leSetAdvertiseEnable(0x01);`
to
`if (ScanResponseDataSet) HCI.leSetAdvertiseEnable(0x01);`

regards,
Paul

cansimkp, 2021-04-11:
Thank you for your prompt reply.

But sorry, your solution doesn't work.
As soon as the following condition is not met, the Nano goes into bootloader mode.
``
 while (peripheral.connected ()) {
    // while the peripheral is connected
    if (bChanged)
    {
      bChanged = 0;
#ifdef USE_OLED
      ShowControllerState (u16Buttons); // update OLED graphic display
#else
      sprintf (ucTemp, "Buttons changed: 0x% 04x", u16Buttons);
      ShowMsg (ucTemp, 0);
#endif
    }
  } 
``

I thought about breaking the while loop after 30 seconds before remote controll breaking the connection with 
"peripheral.disconnect ();"
But here, too, the nano hangs up.

BLE.end (); does not crash the nano. This only happens when the connection is re-established.

best regards,
kp

[HCI.zip](https://github.com/arduino-libraries/ArduinoBLE/files/6292499/HCI.zip)

facchinm, 2021-04-12:
Hi @cansimkp,
the "led blinking mode" signals when the core has crashed. You can get some information of what happened by connecting a serial-to-usb converter to pin TX1 configuring the port as 115200@8n1 ; pasting here the output would greatly help debugging the issue.

Polldo, 2021-04-12:
Hi @cansimkp 
This seems related to this issue https://github.com/arduino-libraries/ArduinoBLE/issues/128
This [PR](https://github.com/arduino/ArduinoCore-mbed/pull/159) on the mbed core should fix it. Can you try this?
thanks

facchinm, 2021-04-12:
The PR should be already included in core 2.0.0, so if the core posted by @Polldo works and it still crashes on 2.0.0 we need to investigate the build process.

cansimkp, 2021-04-12:
@facchinm: with Arduino mbed -enabled Boards V2.0.0:
if connection is lost, here the serial output on TX1:
`
++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0   : 00000000
R1   : 400019A8
R2   : 00000000
R3   : 400183E4
R4   : 7096F5A0
R5   : 00016A3C
R6   : 20000718
R7   : 20000CAC
R8   : 00010261
R9   : 00000000
R10  : 00000000
R11  : 00000000
R12  : BFCEF7FF
SP   : 2000A740
LR   : 20000CB4
PC   : 00053160
xPSR : 210F0000
PSP  : 2000A6D8
MSP  : 2003FFC0
CPUID: 410FC241
HFSR : 40000000
MMFSR: 00000000
BFSR : 00000004
UFSR : 00000000
DFSR : 00000000
AFSR : 00000000
Mode : Thread
Priv : Privileged
Stack: PSP

-- MbedOS Fault Handler --

++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
Location: 0x53160
Error Value: 0x20010764
Current Thread: main Id: 0x2000A95C Entry: 0x19327 StackSize: 0x8000 StackMem: 0x20002938 SP: 0x2000A740
For more info, visit: https://mbed.com/s/error?error=0x80FF013D&osver=60900&core=0x410FC241&comp=2&ver=100200&tgt=ARDUINO_NANO...

-- MbedOS Error Info --
`

facchinm, 2021-04-12:
@cansimkp you can use this utility (https://github.com/janjongboom/mbed-os-example-fault-handler) to decode where the fault happened, given you provide the path to the compiled elf file (you can get it from the verbose output, it's in the root of the build folder). We'll try to replicate too, but it looks specific to the controller / code combination.

cansimkp, 2021-04-12:
After many hours of googling, I was able to generate something.
That didn't work: https://github.com/janjongboom/mbed-os-example-fault-handler
But this one: https://github.com/ARMmbed/mbed-os/tree/master/tools/debug_tools/crash_log_parser
Is that important?

Here the log from debugtool:

I've zipped and attached the entire arduino Build.
And then the output of the compiler console.

C:\Users\nplmk\AppData\Local\Temp\arduino_build_184227>python crash_log_parser.py crash_log.txt C:\Users\nplmk\AppData\Local\Temp\arduino_build_184227\Nano_33_Gamepad.ino.elf C:\Users\nplmk\AppData\Local\Temp\arduino_build_184227\Nano_33_Gamepad.ino.map

Crash Info:
        Crash location = _free_r [0x000530BC] (based on PC value)
        Caller location = __exidx_start [0x20000CB4] (based on LR value)
        Stack Pointer at the time of crash = [2000A740]
        Target and Fault Info:
                Processor Arch: ARM-V7M or above
                Processor Variant: C24
                Forced exception, a fault with configurable priority has been escalated to HardFault
                Imprecise data access error has occurred

C:\Users\nplmk\AppData\Local\Temp\arduino_build_184227>

Console:

[arduino_build_184227.zip](https://github.com/arduino-libraries/ArduinoBLE/files/6299555/arduino_build_184227.zip)
C:\Users\nplmk\Dropbox\Arduino_projekte\Nano_33_Gamepad\Nano_33_Gamepad.ino: In function 'void setup()':
C:\Users\nplmk\Dropbox\Arduino_projekte\Nano_33_Gamepad\Nano_33_Gamepad.ino:76:37: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   ShowMsg("Scanning for BLE ctrl", 0);
                                     ^
C:\Users\nplmk\Dropbox\Arduino_projekte\Nano_33_Gamepad\Nano_33_Gamepad.ino:81:38: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     ShowMsg("starting BLE failed!", 0);
                                      ^
C:\Users\nplmk\Dropbox\Arduino_projekte\Nano_33_Gamepad\Nano_33_Gamepad.ino: In function 'void HIDReportWritten(BLEDevice, BLECharacteristic)':
C:\Users\nplmk\Dropbox\Arduino_projekte\Nano_33_Gamepad\Nano_33_Gamepad.ino:131:11: warning: unused variable 'i' [-Wunused-variable]
 int iLen, i;
           ^
C:\Users\nplmk\Dropbox\Arduino_projekte\Nano_33_Gamepad\Nano_33_Gamepad.ino: In function 'void monitorActions(BLEDevice)':
C:\Users\nplmk\Dropbox\Arduino_projekte\Nano_33_Gamepad\Nano_33_Gamepad.ino:147:29: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   ShowMsg("Connecting...", 4);
                             ^
C:\Users\nplmk\Dropbox\Arduino_projekte\Nano_33_Gamepad\Nano_33_Gamepad.ino:149:27: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     ShowMsg("Connected", 5);
                           ^
C:\Users\nplmk\Dropbox\Arduino_projekte\Nano_33_Gamepad\Nano_33_Gamepad.ino:151:36: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     ShowMsg("Failed to connect!", 5);
                                    ^
C:\Users\nplmk\Dropbox\Arduino_projekte\Nano_33_Gamepad\Nano_33_Gamepad.ino:158:35: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     ShowMsg("0x1812 discovered", 6);
                                   ^
C:\Users\nplmk\Dropbox\Arduino_projekte\Nano_33_Gamepad\Nano_33_Gamepad.ino:160:36: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     ShowMsg("0x1812 disc failed", 6);
                                    ^
C:\Users\nplmk\Dropbox\Arduino_projekte\Nano_33_Gamepad\Nano_33_Gamepad.ino:187:22: warning: converting to non-pointer type 'int' from NULL [-Wconversion-null]
   if (protmodChar != NULL)
                      ^~~~
C:\Users\nplmk\Dropbox\Arduino_projekte\Nano_33_Gamepad\Nano_33_Gamepad.ino:221:29: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   ShowMsg("Ready to use!", 7);
                             ^
C:\Users\nplmk\Dropbox\Arduino_projekte\Nano_33_Gamepad\Nano_33_Gamepad.ino:141:8: warning: unused variable 'iLen' [-Wunused-variable]
 int i, iLen, iCount;
        ^~~~
C:\Users\nplmk\Dropbox\Arduino_projekte\libraries\arduino_955387\src\utility\ATT.cpp: In member function 'virtual bool ATTClass::handleNotify(uint16_t, const uint8_t*, int)':
C:\Users\nplmk\Dropbox\Arduino_projekte\libraries\arduino_955387\src\utility\ATT.cpp:522:39: warning: variable length array 'notification' is used [-Wvla]
     uint8_t notification[_peers[i].mtu];
                                       ^
C:\Users\nplmk\Dropbox\Arduino_projekte\libraries\arduino_955387\src\utility\ATT.cpp: In member function 'virtual bool ATTClass::handleInd(uint16_t, const uint8_t*, int)':
C:\Users\nplmk\Dropbox\Arduino_projekte\libraries\arduino_955387\src\utility\ATT.cpp:552:37: warning: variable length array 'indication' is used [-Wvla]
     uint8_t indication[_peers[i].mtu];
[... 79 lines cut ...]
In file included from C:\Users\nplmk\Documents\ArduinoData\packages\arduino\hardware\mbed\2.0.0\cores\arduino/mbed/rtos/include/rtos/rtos.h:30:0,
                 from C:\Users\nplmk\Documents\ArduinoData\packages\arduino\hardware\mbed\2.0.0\cores\arduino/mbed/mbed.h:24,
                 from C:\Users\nplmk\Documents\ArduinoData\packages\arduino\hardware\mbed\2.0.0\cores\arduino/mbed.h:14,
                 from C:\Users\nplmk\Dropbox\Arduino_projekte\libraries\arduino_955387\src\utility\HCICordioTransport.cpp:23:
[... cut ...]
Globale Variablen verwenden 69352 Bytes (26%) des dynamischen Speichers, 192792 Bytes für lokale Variablen verbleiben. Das Maximum sind 262144 Bytes.
Device       : nRF52840-QIAA
Version      : Arduino Bootloader (SAM-BA extended) 2.0 [Arduino:IKXYZ]
Address      : 0x0
Pages        : 256
Page Size    : 4096 bytes
Total Size   : 1024KB
Planes       : 1
Lock Regions : 0
Locked       : none
Security     : false
Erase flash

Done in 0.001 seconds
Write 316520 bytes to flash (78 pages)
[==============================] 100% (78/78 pages)
Done in 13.326 seconds

Polldo, 2021-04-13:
@cansimkp sorry, I initially misunderstood the problem, it's not related to the issue I linked to.

Can you enable the BLE serial debug and report here the result of the serial monitor?
To enable the BLE debug you can simply add the following line in the setup function:
`BLE.debug(Serial)`

Also, you could try to insert this three-line patch into your ArduinoBLE library: https://github.com/arduino-libraries/ArduinoBLE/pull/96/commits/980df5ae597b62422bc96e5db06c1e67bca55cab#diff-1a79f029352e98b813aaf6a97ee51cf595503c6e75ae51cb4a04a5f705db9853R261-R263
Then let me know if anything changes

Thank you

cansimkp, 2021-04-13:
@Polldo:
BLE.debug(Serial) shows this:

20:19:39.091 -> Found 42:6a:a9:66:5f:9b
20:19:39.091 ->  - fd6f
20:19:39.091 -> HCI EVENT RX <- 043E1002010201E72545A4AC510403036FFDBB
20:19:39.091 -> HCI EVENT RX <- 043E0C02010401E72545A4AC5100BB
20:19:39.091 -> Found ff:ff:38:5a:d2:4e
20:19:39.091 -> Magicsee R1        - 1812                            **<-- Remote connected**
20:19:39.091 -> HCI COMMAND TX -> 010C20020000
20:19:39.091 -> HCI EVENT RX <- 043E2802010201B8D3AAE8DE7E1C03036FFD17166FFD943BFBEFAC784F07388AE1CF86A21E60C5CFDDBAB6
20:19:39.124 -> HCI EVENT RX <- 040E04010C2000
20:19:39.124 -> Connecting...
20:19:39.124 -> HCI COMMAND TX -> 010D20196000300000004ED25A38FFFF0006000C000000C80004000600
20:19:39.160 -> HCI EVENT RX <- 040F0400010D20
20:19:39.160 -> HCI EVENT RX <- 043E130100000000004ED25A38FFFF08000000C80007
20:19:39.160 -> Connected
20:19:39.160 -> HCI ACLDATA TX -> 02000007000300040002F700
20:19:39.200 -> HCI EVENT RX <- 0413050104000100
20:19:39.200 -> HCI EVENT RX <- 0413050100000100
20:19:39.200 -> HCI ACLDATA RX <- 020020070003000400031700
20:19:39.200 -> HCI ACLDATA TX -> 0200000B0007000400100100FFFF0028
20:19:39.241 -> HCI EVENT RX <- 0413050104000100
20:19:39.241 -> HCI EVENT RX <- 0413050100000100
20:19:39.241 -> HCI ACLDATA RX <- 020020180014000400110601000700001808000B0001180C000E000A18
20:19:39.286 -> HCI ACLDATA TX -> 0200000B0007000400100F00FFFF0028
20:19:39.286 -> HCI EVENT RX <- 0413050104000100
20:19:39.286 -> HCI EVENT RX <- 0413050100000100
20:19:39.286 -> HCI ACLDATA RX <- 02002012000E00040011060F002700121828002A000F18
20:19:39.321 -> HCI ACLDATA TX -> 0200000B0007000400102B00FFFF0028
20:19:39.360 -> HCI EVENT RX <- 0413050104000100
20:19:39.360 -> HCI EVENT RX <- 0413050100000100
20:19:39.360 -> HCI ACLDATA RX <- 02002009000500040001102B000A
20:19:39.360 -> HCI ACLDATA TX -> 0200000B0007000400080F0027000328
20:19:39.400 -> HCI EVENT RX <- 0413050104000100
20:19:39.400 -> HCI EVENT RX <- 0413050100000100
20:19:39.400 -> HCI ACLDATA RX <- 0200201B0017000400090710000611004E2A12001213004D2A16001217004D2A
20:19:39.435 -> HCI ACLDATA TX -> 0200000B000700040008180027000328
20:19:39.476 -> HCI EVENT RX <- 0413050104000100
20:19:39.476 -> HCI EVENT RX <- 0413050100000100
20:19:39.476 -> HCI ACLDATA RX <- 0200201B001700040009071A00121B004D2A1E00121F004D2A22000223004B2A
20:19:39.476 -> HCI ACLDATA TX -> 0200000B000700040008240027000328
20:19:39.516 -> HCI EVENT RX <- 0413050104000100
20:19:39.516 -> HCI EVENT RX <- 0413050100000100
20:19:39.516 -> HCI ACLDATA RX <- 020020140010000400090724000225004A2A26000427004C2A
20:19:39.550 -> HCI ACLDATA TX -> 0200000B000700040008280027000328
20:19:39.550 -> HCI EVENT RX <- 0413050104000100
20:19:39.550 -> HCI EVENT RX <- 0413050100000100
20:19:39.590 -> HCI ACLDATA RX <- 020020090005000400010828000A
20:19:39.590 -> HCI ACLDATA TX -> 0200000900050004000412001300
20:19:39.590 -> HCI EVENT RX <- 0413050104000100
20:19:39.590 -> HCI EVENT RX <- 0413050100000100
20:19:39.624 -> HCI ACLDATA RX <- 0200200E000A00040005011200032813004D2A
20:19:39.624 -> HCI ACLDATA TX -> 0200000900050004000414001300
20:19:39.624 -> HCI EVENT RX <- 0413050104000100
20:19:39.624 -> HCI EVENT RX <- 0413050100000100
20:19:39.664 -> HCI ACLDATA RX <- 020020090005000400010414000A
20:19:39.664 -> HCI ACLDATA TX -> 0200000900050004000414001700
20:19:39.664 -> HCI EVENT RX <- 0413050104000100
20:19:39.700 -> HCI EVENT RX <- 0413050100000100
20:19:39.700 -> HCI ACLDATA RX <- 020020160012000400050114000229150008291600032817004D2A
20:19:39.700 -> HCI ACLDATA TX -> 0200000900050004000418001700
20:19:39.746 -> HCI EVENT RX <- 0413050104000100
20:19:39.746 -> HCI EVENT RX <- 0413050100000100
20:19:39.746 -> HCI ACLDATA RX <- 020020090005000400010418000A
20:19:39.746 -> HCI ACLDATA TX -> 0200000900050004000418001B00
20:19:39.787 -> HCI EVENT RX <- 0413050104000100
20:19:39.787 -> HCI EVENT RX <- 0413050100000100
20:19:39.787 -> HCI ACLDATA RX <- 020020160012000400050118000229190008291A0003281B004D2A
20:19:39.787 -> HCI ACLDATA TX -> 020000090005000400041C001B00
[... 53 lines cut ...]
20:19:40.364 -> HCI EVENT RX <- 0413050104000100
20:19:40.400 -> HCI EVENT RX <- 0413050100000100
20:19:40.400 -> HCI ACLDATA RX <- 02002005000100040013
20:19:40.400 -> HCI ACLDATA TX -> 02000008000400040052110001
20:19:40.400 -> Ready to use!
20:19:42.390 -> HCI EVENT RX <- 0413050104000100
20:19:42.431 -> HCI EVENT RX <- 0413050100000100
20:19:42.431 -> Buttons changed: 0x0000
20:19:43.736 -> HCI ACLDATA RX <- 0200200900050004001B17000100
20:19:43.776 -> Buttons changed: 0x0501
20:19:43.936 -> HCI ACLDATA RX <- 0200200900050004001B17000000
20:19:43.936 -> Buttons changed: 0x0500
20:19:44.644 -> HCI ACLDATA RX <- 0200200900050004001B17000200
20:19:44.684 -> Buttons changed: 0x0502
20:19:44.862 -> HCI ACLDATA RX <- 0200200900050004001B17000000
20:19:44.862 -> Buttons changed: 0x0500
20:19:50.482 -> HCI EVENT RX <- 04050400000008
20:19:50.482 -> HCI EVENT RX <- 0413050100000000                **at this point i disconnet remote control and nano crashed**

The nano also crashes with the three line patch.

But I found something out.
When I upload the modified LED example, nothing crashes. However, this only searches for the battery service 0x180F and the characteristic 0x2A19. Unfortunately, my BLE knowledge is not sufficient to establish the HID connection. That's why I wanted to take over the part from bitbank2.

best regards kp

cansimkp, 2021-04-13:
I changed this to this:

` **DELETED** BLEService hidService;

  if (peripheral.connect()) {
    Serial.println("Connected");
  } else {
    Serial.println("Failed to connect!");
    return;
  }

  if (peripheral.discoverService("1812")) {
    Serial.println("0x1812 discovered");
  } else {
    Serial.println("0x1812 disc failed");
    peripheral.disconnect();

   **DELETED** while (1);
    return;
  }
**added** BLEService hidService     = peripheral.service("1812"); // get the HID service
**DELETED**  hidService = peripheral.service("1812"); // get the HID service`

NOW IT RUN !!!!

Thank u all

closed 2021-04-13 by cansimkp