arduino-libraries/ArduinoBLE #175: Problem with BLEDevice.discoverAttributes in 1.2.0 version

akumar36, 2021-04-21:
I'm working on Arduino Nano 33 IoT and trying to discover attributes of nearby BLE devices. 
With BLE library version 1.1.3, everything works fine. But, with version 1.2.0, the Arduino device goes to a bad state when I execute discover attributes for some BLE peripherals(Mi band in this case).

I'm attaching my sketch which scans nearby devices and discovers their attributes.
[BLE_Explorer.ino.zip](https://github.com/arduino-libraries/ArduinoBLE/files/6349351/BLE_Explorer.ino.zip)

baconcheese113, 2021-05-23:
Running into the same issue, have you found a workaround yet?

My logs from nRF Connect:

```
22:10:39>Connecting to <MAC_ADDRESS>
22:10:39>gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
22:10:39>[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
22:10:39>Connected to <MAC_ADDRESS>
22:10:39>Discovering services...
22:10:39>gatt.discoverServices()
22:10:39>[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
22:10:40>Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
22:10:54>[Callback] Connection state changed with status: 22 and new state: DISCONNECTED (0)
22:10:54>Error 22 (0x16): GATT CONN TERMINATE LOCAL HOST
22:10:54>Disconnected
22:10:54>[Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
```
It's also worth mentioning that when it stops advertising, both the OnConnect and OnDisconnect handlers are called

baconcheese113, 2021-05-23:
I isolated my issue down to `BLE.advertise()` not also allowing service discovery by peripherals

Not sure why, but calling `BLE.available()` every loop cycle allowed services to be discovered

akumar36, 2021-06-23:
The issue seems to be due to this submission: https://github.com/arduino-libraries/ArduinoBLE/commit/3074107d6ea9f9d85825acf90219987ed8dd793e?branch=3074107d6ea9f9d85825acf90219987ed8dd793e&diff=split

Reverting the above change back resolves the issue.

akumar36, 2021-06-23:
> I isolated my issue down to `BLE.advertise()` not also allowing service discovery by peripherals
> 
> Not sure why, but calling `BLE.available()` every loop cycle allowed services to be discovered

calling BLE.available() function did not help me.

ElreboCM, 2021-08-13:
I have the same issue on Arduino BLE Nano 33 with a  Cycling Power Meter.

For Reference I have attached debug dumps for 
[hcidump-bsp1.3.1-library1.1.3.txt](https://github.com/arduino-libraries/ArduinoBLE/files/6983639/hcidump-bsp1.3.1-library1.1.3.txt)
[hcidump-bsp2.4.1-library1.2.1.txt](https://github.com/arduino-libraries/ArduinoBLE/files/6983640/hcidump-bsp2.4.1-library1.2.1.txt)

- Arduino Nano 33 BSP 1.3.1 - ArduinoBLE 1.1.3 (working)
- Arduino Nano 33 BSP 2.4.1 - ArduinoBLE 1.3.1 (with the bug described above)

ElreboCM, 2021-08-14:
I can confirm that reverting the change of this [commit](https://github.com/arduino-libraries/ArduinoBLE/commit/3074107) resolves the issue.

TheMadMoose, 2023-03-06:
Small update just in case someone else was experiencing this issue as of March 2023: Reverting the change of the aforementioned commit resolved it for me as well (Hardware: Portenta H7 connecting over BLE to two Nicla Sense ME boards).

Elaboration on our setup:
Both Nicla boards have a unique BLE device name, but identical service and characteristic UUIDs. 
We cycle through each peripheral (nicla) on the Central device. The first Nicla was discovered as expected, however the second one kept failing the characteristic discovery step.

Changing "J+1" to "J" seemed to immediately resolve this.

If I broke any rules by commenting / updating - My sincere apologies :)

ForrestThump, 2024-02-01:
@TheMadMoose Can you elaborate on where I should change J+1 to J? My peripheral discoverAttributes() call always returns false, and I'm at a loss. I'm using two esp32s...