arduino-libraries/ArduinoBLE #179: How to change BLE address?

Klauzen, 2021-05-22:
Hi,

im using this library with my arduino portenta's, but i need to change the preset address (AA:AA:AA:AA:AA:AA) to a custom address... Is this possible with this library?

Thanks,
Klaus

facchinm, 2021-05-24:
Hi @Klauzen ,
which version of the Portenta core are you using? Any core we released officially should contain [this](https://github.com/arduino/ArduinoCore-mbed/blob/master/patches/0014-portentaH7-Change-BLE-public-address.patch) patch that unsets the default BLE address and replaces with our own.

Klauzen, 2021-05-24:
Hey @facchinm ,

thank you so much! I've updated the core version to 2.1.0 and now it works perfectly!
Best regards,
Klaus

closed 2021-05-24 by facchinm