arduino-libraries/ArduinoBLE #190: Unable to retrieve UUID in setEventHandler's callback

glsubri, 2021-07-08:
Hello everyone.

I am using the ArduinoBLE library on my Nano 33 BLE.

I have a class, `DayAlarmService` which has three characteristics: `enabled`, `startTime` and `duration`. I have 7 instances of that class, one for each day of the week. I need to know on which one of them an event was triggered (like a read on a characteristic or a subscription, ...).

To do this, I have another class, `AlarmServiceManager`, which instantiates all 7 `DayAlarmServices` and stores all UUIDs (from the services and the characteristics). Since we have access to the `BLECharacteristic` when we set an event handler, I should be able to retrieve the `uuid` of the characteristic and check out to which day it corresponds. Of course, all characteristics and service have a different UUID, which are known by `AlarmServiceManager`.

The issue is that, for an unknown reason, when the callback is called, the `uuid()` function returns "random characters" and not the correct uuid. An example of the ouput is:
  
[image]

To make things easier, I have set up a few gists for you to view the code in question:

https://gist.github.com/glsubri/d5ed63afc946632d096ce7087aedc818
https://gist.github.com/glsubri/98516dad7d289ac1b0db14d35a6cf99d

This file contains multiple UUIDs definition:
https://gist.github.com/glsubri/f55c598d14261c36c40e17c20f87714a

The weird part is that I have other services and characteristics that can correctly access their `uuid()`. I don't understand why it doesn't work as expected in this case.

I hope that I was clear. If you need anything else, please let me know!

Thanks

P.S. I also have written something in Arduino's forum: https://forum.arduino.cc/t/unable-to-retrieve-characteristic-uuid-in-callback/882635

polldo, 2021-07-08:
Hi @glsubri 
The problem should be that your characteristics' UUIDs are allocated on the stack [here](https://gist.github.com/glsubri/98516dad7d289ac1b0db14d35a6cf99d#file-alarmservicemanager-cpp-L21-L25)
You could either give them a global scope, like you did for services' UUIDs, or allocate them on to the heap

glsubri, 2021-07-08:
@polldo Oh, I see. I thought the uuids would be copied inside of the `BLECharacteristic` and `BLEService`.
I will probably store them in my `AlarmServiceManager`.

Thank you very much. I will get back to you when everything is working !

glsubri, 2021-07-08:
So, everything seems to be fine when the uuids are stored statically.

However, wouldn't it make more sense if the `BLECharacteristic` and `BLEService` would be in charge of storing their uuid themselves (and copying it) instead of relying on other objects ?

polldo, 2021-07-08:
That's a good point, can you make another issue to request this feature?
I'm going to close this one as it is solved.

closed 2021-07-08 by polldo