arduino-libraries/ArduinoBLE #192: ArduinoBLE crashes when BLE.begin BLE.end is called multiple times (possible solution included)

Klaus-KK, 2021-07-14:
The issue was reported by a user in the Arduino forum, who wants to switch between BLE and WiFi on a regular basis.

[https://forum.arduino.cc/t/repeated-ble-begin-end-crashes-board/885009](https://forum.arduino.cc/t/repeated-ble-begin-end-crashes-board/885009)

I wrote the following sketch for testing and it crashes after 50 loops.

```
#include <ArduinoBLE.h>

void setup()
{
  Serial.begin( 9600 );
  while ( !Serial );
  Serial.println( "ArduinoBLE memory leak test" );
}

void loop()
{
  static uint32_t counter = 0;

  BLE.begin();
  counter++;
  Serial.print( "C: " );
  Serial.println( counter );
  BLE.end();
}

```

With my limited C++ knowledge I found in void GATTClass::begin() a couple of objects are created with new. They are not deleted in void GATTClass::end().

Adding the 5 deletes to GATTClass::end seems to fix the issue. Can you confirm this is the right way to fix this and implement this in the next version of the library?

```
void GATTClass::end()
{
  delete( _genericAccessService );
  delete( _deviceNameCharacteristic );
  delete( _appearanceCharacteristic );
  delete( _genericAttributeService );
  delete( _servicesChangedCharacteristic );
 _attributes.clear();
}
```

sneha-dsp, 2021-08-23:
I am getting same issue as @Klaus-KK my Arduino Nano 33 BLE gets stuck in either BLE.begin() or BLE.discoverattributes() and never recovers from it unless a hard reset.

MCXIV, 2021-11-26:
Hi,

Same issue for me with MKR 1010. It crashes at loop n°45. Your workaround @Klaus-KK works for me, thanks. 😊

Quentin

ricozinn, 2022-04-20:
Same crash for me with the Artemis Apollo3.  Can that fix get pulled in to an official release?

alranel, 2022-05-12:
I created a pull request containing this fix:
https://github.com/arduino-libraries/ArduinoBLE/pull/237

buffcode, 2022-05-20:
No leak in 1.1.3, but starting with 1.2.0 most likely due to https://github.com/arduino-libraries/ArduinoBLE/commit/de392e0b96ff2210a2bf76a657ca54aca045123f

ricozinn, 2022-05-20:
> No leak in 1.1.3, but starting with 1.2.0 most likely due to [de392e0](https://github.com/arduino-libraries/ArduinoBLE/commit/de392e0b96ff2210a2bf76a657ca54aca045123f)

I just tried undoing those changes you referenced and on the Artemis chip it still crashes at the same place (after calling BLE.begin() followed by BLE.end() 26 times.

buffcode, 2022-05-20:
> I just tried undoing those changes you referenced and on the Artemis chip it still crashes at the same place (after calling BLE.begin() followed by BLE.end() 26 times.

@ricozinn Hm okay, did you also try to use 1.1.3 completely? I have no memory leak with that version, but maybe the referenced commit is not (fully) responsible.

ricozinn, 2022-05-20:
@alranel I tried your changed in this PR: https://github.com/arduino-libraries/ArduinoBLE/pull/237 with Artemis chip and it fails (same MBedOS hard fault) after only one cycle.  I think the clearAttributes() tries to release memory already released in the lines above it.  

I've tried several variations of your suggested changes and for Artemis they crash after 26 cycles of BLE.begin() followed by BLE.end().

Meaning, your proposed changes have the same net effect of the hard fault in MBedOS as leaving the ::end as this:
```
void GATTClass::end()
{ 
  _attributes.clear();
}
```

I can't use that MemoryUsage library though to see what's happening to memory at each step because it isn't compatible with Artemis.

Thanks for proposing solutions for this, let me know if you have any other ideas?  I'm stumped on a project trying to use ArduinoBLE currently that needs to be able to turn BLE on and off until I can either think of an alternative or find a way to get this working.

ricozinn, 2022-05-20:
@buffcode, unfortunately since 1.1.3 release they added support for the
Apollo3 core (support for Artemis) that I am using, so I can't go all the
way back to that one.

ricozinn, 2022-05-20:
> You are right, the services are part of the attributes and as such are
> already released. So only the characteristics and attributes need to be
> released explicitly, otherwise the board will crash (in my case).
> The memory leak is somewhat reduced but still there and kills the board
> after 20-ish runs.

Ok, glad you are seeing it too.  Let me konw if you discover a fix.
Thanks!  Are you also using the Apollo3 core?  Or some other board?

MCXIV, 2022-05-20:
> @alranel I tried your changed in this PR: https://github.com/arduino-libraries/ArduinoBLE/pull/237 with Artemis chip and it fails (same MBedOS hard fault) after only one cycle.  I think the [...]

Had the same problem, had to use a watchdog to reset the board when this problem occurred.

closed 2022-05-30 by facchinm