arduino-libraries/ArduinoBLE #218: BLE won't reconnect unless serial port opened up

sup4rl33thax0r, 2022-01-11:
1.  Hardware is a Sparkfun Artemis Nano.  If this isn't the right place I apologize.
2. Code connects to peripheral, reads characteristics, etc.
3. If I disconnect peripheral (say by disconnecting power), code refuses to do anything until the serial port in the IDE is opened.  It doesn't need it open to connect the first time, but it seems to hang the second go around until the serial port window is opened back up.  Importantly, there isn't a single call to Serial in the code.  Not even Serial.begin()

Code follows:

`
#include "strings.h"
#include <mbed.h>

#include <Wire.h>
#include <ArduinoBLE.h>
#include <SparkFun_Alphanumeric_Display.h>

HT16K33 display;

uint16_t driver_odo;
char SevenSeg1String[6]; 

int seven_seg_brightness_a;
int seven_seg_brightness_b;
bool ble_connected;

using namespace rtos;
Thread display_thread;
Thread ble_thread;

void setup() {
  ble_connected = false;

  Wire.begin();
  BLE.begin(); 
  BLE.scan();
  display.begin(0x70);

  display_thread.start(update_displays);
  ble_thread.start(manage_ble);
}

void update_displays(){

  while(1){
    
    if(ble_connected == true){
      String dist = String(driver_odo);   
      sprintf(SevenSeg1String, "% 4s", dist.c_str());   
      display.print(SevenSeg1String);
    }else{   
      display.print("----");
    }    
  }
}

void manage_ble(){
  while(1){
    
    BLEDevice peripheral = BLE.available();
    
    if (peripheral) {
      if (peripheral.advertisedServiceUuid() == "59839c23-8109-4ac1-bde9-62269a6550a4") {
        BLE.stopScan();
        manage_ble_connection(peripheral);
      }
    }    
    ble_connected = false;
    BLE.scan();
    delay(100);
    
   }
}
void loop() {

}

void manage_ble_connection(BLEDevice peripheral) {
  // connect to the peripheral
  if (peripheral.connect()) {
    ble_connected = true;
  } else {
    return;
  }

  // discover peripheral attributes
  if (peripheral.discoverService("59839c23-8109-4ac1-bde9-62269a6550a4")) {
  } else {
    peripheral.disconnect();
    
    return;
  }

  // retrieve the simple key characteristic
  BLECharacteristic simpleKeyCharacteristic = peripheral.characteristic("1d1a6dc8-3870-448d-bd1a-3e34f63d397f");

  // subscribe to the simple key characteristic
  if (!simpleKeyCharacteristic) {
    peripheral.disconnect();
    return;
  } else if (!simpleKeyCharacteristic.canSubscribe()) {
    peripheral.disconnect();
    return;
  } else if (!simpleKeyCharacteristic.subscribe()) {
    peripheral.disconnect();
    return;
  } else {
  }

  while (peripheral.connected()) {
    // while the peripheral is connected

    // check if the value of the simple key characteristic has been updated
    if (simpleKeyCharacteristic.valueUpdated()) {
      // yes, get the value, characteristic is 1 byte so use byte value
      byte value = 0;    
      simpleKeyCharacteristic.readValue(driver_odo);
      delay(100);
    }
  }
    Serial.flush();
    BLE.disconnect();
  return;
}`

facchinm, 2022-01-11:
@Wenn0101 can you help on this?

Wenn0101, 2022-01-11:
I suspect that the serial port opening is causing a _system reset_.

@sup4rl33thax0r
Can you try a reset of the Artemis at the same time you would typically open the serial port to see if you get them same result?

I think that this means that the reset of the peripheral is causing problems that can only be fixed by a reset of your central device.  
Taking a look at your sketch, I think the problem might be that you are restarting the scan in every loop. I suspect this would mean that if the first peripheral returned by BLE.available() is not the one you are looking for, you restart the scan and may see the same device again the next time you call BLE.available(). I am not 100% sure if this would cause this behavior, and can do some of my own verification, but can you try this instead?
```
void loop() {
    BLEDevice peripheral = BLE.available();
    
    if (peripheral) {
      if (peripheral.advertisedServiceUuid() == "59839c23-8109-4ac1-bde9-62269a6550a4") {
        BLE.stopScan();
        manage_ble_connection(peripheral);
        BLE.scan();
      }
    }    
    ble_connected = false;
   
}
```

You can even try replacing your ` BLE.scan();` calls with ` BLE.scanForUuid("59839c23-8109-4ac1-bde9-62269a6550a4");

Quick question:
What version of the ArduinoBLE Library and Artemis are you using?

sup4rl33thax0r, 2022-01-12:
Thank you @Wenn0101 for helping!  I think you're right that the Serial console was doing a reset.  I verified that behavior with a different sketch.  Apologies, it's been awhile since I did anything Arduino and I wasn't aware of that behavior.  

I thought you were onto something with the idea of the scan being constantly restarted but making the changes you suggested don't seem to make a difference.

Thank you again for offering your help, I've been fighting with this for awhile and appreciate it so much.

Wenn0101, 2022-01-12:
@sup4rl33thax0r I am not sure if all arduino boards this way, but this reset is a reality of the hardware used in the Sparkfun Artemis Nano.

Given what we know this issue might best be rephrased "BLE won't reconnect to peripheral, unless central is reset", I dug into this a little bit and it seems like the BLE.scan(); seems to be failing after the peripheral is disconnected. Thus, nothing every shows up as available after this point. I checked with a simple sketch and it seems that this hardware has no problem starting and stopping scans. This confirms this is an issue only when the peripheral disconnects, not a general problem with restarting a scan.

Perhaps we can loop back in @facchinm. I verified this issue with the LED, and LedControl examples (2 devices, allow them to connect then reset the peripheral). Can you verify if this is a problem on other hardware or if it is a problem with just the Artemis Boards. If its a problem with Artemis, I'll create an issue in the Artemis repo.