arduino-libraries/ArduinoBLE #222: Nano BLE Corrupts Descriptors if i comment Serial.println

LRagji, 2022-02-27:
Hi team, i am using following program with BLE Nano 33 Sense and there is a weird thing going on if i comment `line 82` which is a serial println one of my format descriptors in the temperature characteristics goes hay wire with some random bits showing up(validate with nrf connect app), it is expected to always stay float ieee-754.. looks like some memory corruption, can some one help?
```
#include <ArduinoBLE.h>
#include <Arduino_HTS221.h>

#define BLEBlocksServiceId "68469193-feb1-4953-920d-8a0953a9229c"
#define TemperatureCharId "68469193-feb2-4953-920d-8a0953a9229c"
#define HumidityCharId "68469193-feb3-4953-920d-8a0953a9229c"
#define UserDescriptorID "2901"
#define PresentationFormatDescriptorID "2904"
#define TemperatureEpsillion 0.20
#define HumidityEpsillion 0.20

// Custom BLE Chars
BLEFloatCharacteristic TempChar(TemperatureCharId, BLERead | BLENotify);
BLEFloatCharacteristic HumidityChar(HumidityCharId, BLERead | BLENotify);

void setup() {
  //Indicate Setup started
  digitalWrite(LED_BUILTIN, HIGH);

  // initialize serial communication
  Serial.begin(9600);
  while (!Serial);

  // initialize Temp/Humidity sensor
  if (!HTS.begin()) {
    Serial.println("TH sensor init failed!!,Program Halted");
    while (1);
  }
  float defaultValue = HTS.readTemperature();
  TempChar.writeValue(defaultValue);
  defaultValue = HTS.readHumidity();
  HumidityChar.writeValue(defaultValue);

  //initialize BLE communication
  if (!BLE.begin()) {
    Serial.println("BLE init failed!!,Program Halted");
    while (1);
  }
  BLEService BLEBlocksService(BLEBlocksServiceId);
  BLE.setAppearance(0x0540);//Generic Sensor
  BLE.setLocalName("BLE-Blocks");
  BLE.setDeviceName("BLE-Blocks");
  BLE.setConnectable(true);
  BLE.setAdvertisingInterval(800);//500ms to save battery
  BLEDescriptor TemperatureUserDescriptor(UserDescriptorID, "Temperature");
  //For 7 byte defination visit https://gist.github.com/heiko-r/f284d95141871e12ca0164d9070d61b4
  // https://github.com/oesmith/gatt-xml/blob/master/org.bluetooth.descriptor.gatt.characteristic_presentation_format.xml
  //For Enums of each type visit https://os.mbed.com/docs/mbed-os/v6.15/mbed-os-api-doxy/struct_gatt_characteristic_1_1_presentation_format__t.html#ae556940a3f863dbd5ffd135801ade996
  //7 byte divided into 5 parts format, exponent, unit, name space and description
  byte DegreeCelsiusGATTUnit[] = {0x14, 0x00, 0x2F, 0x27, 0x01, 0x00, 0x00};
  BLEDescriptor TemperatureFormatDescriptor(PresentationFormatDescriptorID, DegreeCelsiusGATTUnit, 7);
  TempChar.addDescriptor(TemperatureUserDescriptor);
  TempChar.addDescriptor(TemperatureFormatDescriptor);
  BLEDescriptor HumidityUserDescriptor(UserDescriptorID, "Humidity");
  byte PercentageGATTUnit[] = {0x14, 0x00, 0xAD, 0x27, 0x01, 0x00, 0x00};
  BLEDescriptor HumidityFormatDescriptor(PresentationFormatDescriptorID, PercentageGATTUnit, 7);
  HumidityChar.addDescriptor(HumidityUserDescriptor);
  HumidityChar.addDescriptor(HumidityFormatDescriptor);
  BLEBlocksService.addCharacteristic(TempChar);
  BLEBlocksService.addCharacteristic(HumidityChar);
  BLE.setAdvertisedService(BLEBlocksService);
  BLE.addService(BLEBlocksService);
  BLE.advertise();

  //Indicate Setup completed
  digitalWrite(LED_BUILTIN, LOW);
}

void loop() {
  BLEDevice client = BLE.central();
  if (client) {
    while (client.connected())
    {
      float temperature = 1.0;//HTS.readTemperature();
      float humidity    = 2.0;//HTS.readHumidity();
      if (TempChar.subscribed())
      {
        float exisitingTemp = 2.0;
        Serial.println(TempChar.value());
        Serial.println(HumidityChar.value());
        exisitingTemp = exisitingTemp - temperature;
        Serial.println("T: " + String(abs(TempChar.value() - temperature)) + " H: " + String(abs(HumidityChar.value() - humidity)) + String(humidity));
        if (abs(exisitingTemp) > float(TemperatureEpsillion))
        {
          TempChar.writeValue(temperature);
        }
      }
      if (HumidityChar.subscribed())
      {
        float existingHumidity = 3.0;//HumidityChar.value();
        existingHumidity = existingHumidity - humidity;
        if (abs(existingHumidity) > float(HumidityEpsillion))
        {
          HumidityChar.writeValue(humidity);
        }
      }
      delay(100);
    }
  }
  delay(100);
}
```

LRagji, 2022-02-27:
Quick workaround with this one if i declare the descriptors globally then everything works, some serious C memory woo-do in happening in the library...

facchinm, 2022-02-28:
Hi @LRagji ,
this is totally expected, since when you declare something like `BLEDescriptor HumidityFormatDescriptor(PresentationFormatDescriptorID, PercentageGATTUnit, 7);` inside `setup()` ,  it goes out of scope just after the function it's declared into finishes (its memory gets "recycled").

To overcome this, you can either
* declare it outside any function (global)
* declare as `static` inside the function
* declare with `new` and pass its pointer <- not very arduinish

LRagji, 2022-02-28:
@facchinm  Isn't this bad coding practice?(or i am expecting high level language patterns here) I mean i gave Characteristics an instance and the Characteristics is global should it not hold on to its reference until Characteristics is gone out of scope?

facchinm, 2022-02-28:
Not really in c++, since every time you create an object this way it gets allocated in the stack and the stack gets trashed as soon as the function finishes, no matter if it's referenced by others.
There isn't really a way to make c++ behave as higher level languages (like python or go) since there's no garbage collector, and the memory allocated would be lost forever (there's no way to recover the address of the object being trashed)