arduino-libraries/ArduinoBLE #227: BLE events handling via I

YannickMarietti, 2022-03-28:
Hello

This is not really an issue but rather a comment on a possible improvement.
In current implementation, ArduinoBLE is managing events via polling :

```
// poll for BLE events
BLE.poll();
```
This can be an issue because you can experience delays on BLE communication if the ` loop() ` function is managing a time-consuming code before / after the `BLE.poll()` call.

The only events currently handled as interrupts are for connected/disconnected: https://www.arduino.cc/en/Reference/ArduinoBLEBLEsetEventHandler.

Is it possible to manage **all BLE events** by interrupts, for example triggering an IT attached to a given characteristic when a central take an action on it ?

Best regards

Yannick

KMeldgaard, 2022-05-05:
Hi @YannickMarietti and rest

I've run into the same issue, not wanting to "spam" BLE.poll(), since my MCU goes into low power mode in the idle task (STM32FreeRTOS).
At the moment, I have to call BLE.poll() every 1ms when the device is connected - of course I could turn that down, but then the BLE response time becomes slower...

Depending on your MCU (in my case a STM32WB) some function is fired every time there is BLE activity. ultimately writing the BLE packet into _rxBuf (or whatever its called for the specific HCI-transport class).
I've played around with having that function trigger i task that calls BLE.poll(), but with no luck, as the MCU faults.
On the STM32WB the recieve callback is `IPCC_C1_RX_IRQHandler`

Maybe this gives some pointers on where to look??
Would be awesome if a solution is proposed..

Regards.

mklemarczyk, 2022-08-07:
I have similar issue, but on the receiver side.
I want to put my module to sleep at night and not process the new values on the characteristics. As the Bluetooth is hosted on separate CPU, could I ignore all interrupts and put my Nano in sleep mode?
Than in the morning the module will handle the last new value send to BLE.