arduino-libraries/ArduinoBLE #241: Pairing Started and Disconnected in Arduino BLE Sense, But Why???

laviator98, 2022-06-09:
Recently Arduino team also added this Lib in their Page as V 1.3.1 and 1.3.0 . I uploaded the Encrypted Battery Monitor Example in my Arduino 33 BLE Sense nrf52840. First, the Numeric Comparison was done and 6 Digit Pin is passed to the mobile phone. By clicking Pair Button, it got disconnected. I also Checked the Scripts, am i missing something?? . Kindly someone help me here. Does Arduino BLE Sense Supports Pairing Process???

Also Adafruit BLE lib (BlueFruit52) also Contains Pairing Examples.
https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/Bluefruit52Lib/examples/Peripheral/pairing_passkey

But Why Arduino is lagging to give a pairing Example but Adafruit already gave Pairing Examples in their Lib.

mklemarczyk, 2022-06-24:
This library does not support paring at the current state.

per1234, 2022-06-27:
> Why Arduino is lagging to give a pairing Example

As you are already aware, a pairing example was added at the same time as the pairing support (https://github.com/arduino-libraries/ArduinoBLE/pull/156).

laviator98, 2022-06-27:
> As you are already aware, a pairing example was added at the same time as the pairing support (#156).

@per1234 Are You Mentioning the Encrypted Battery Monitor Example? It is not working Bro. I think the code is not completed correctly. I am Using android phone for pairing and it is showing the error as "Pass Key or PIN is incorrectly Entered". Also I can't find the Pass Key Submission in this Lib.

Also In Adafruit Bluefruit52 Lib, they used nrf52, S140 API for the Pairing and Encryption.
https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/cores/nRF5/nordic/softdevice/s140_nrf52_6.1.1_API/include

I don't know why for Arduino this S140 API is not used?

I think this is the proper solution for the Paring and Encryption. I'm Also Working on it. Kindly please everyone work on it.

mklemarczyk, 2022-07-02:
I remember that there were two pairing methods used in the past. One with hidden key and one with open key. In the first one the key had to be communicated to the host by other means, like screen or Serial and entered on the host device. The second one was just to confirm the pairing key is the same as in pairing device and it was displayed to the user to confirm that this is device that you are connecting to.

In any way I believe that you should be able to get your pairing key from the Arduino by just printing it to console or display on the 8 digit display.

Jesse-AH-Carpenter, 2022-07-06:
I've been running into a same problem. When using the EncryptedBatteryMonitor example, it displays the mac address of the device that is attempted to pair with the Arduino device, but fails to create a key, and proceeds to disconnect from the device. I've tried all of the scanner apps they recommend and none of them work.

laviator98, 2022-07-07:
As I go through the scripts of ArduinoBLE V 1.3.1 and Adafruit Bluefruit52 Libs, it shows that Arduino BLE uses this docs for connection and pairing mode:
https://www.bluetooth.com/specifications/specs/core-specification-4-2/
https://www.bluetooth.org/docman/handlers/downloaddoc.ashx?doc_id=441541

Adafruit uses this API from nrf site for pairing for only a certain APP of Adafruit or nrf Connect:
https://www.nordicsemi.com/Products/Development-software/S140/Download?lang=en#infotabs

laviator98, 2022-07-08:
In Arduino BLE Lib, by enabling (uncommenting) BLE_TRACE in scripts L2CAPSignaling.cpp, HCI.cpp, ATT.cpp we will get Internal data for Communication.

All Stages for pairing is done but it is not working,
Kindly someone help me here.

[BLE OUT.txt](https://github.com/arduino-libraries/ArduinoBLE/files/9070957/BLE.OUT.txt)

alexpopof, 2022-07-08:
I am seeing similar behavior.  For what it's worth, I recall testing around October 2021 when I saw activity in #156 (@unknownconstant).  Things worked well and it was also tested by @alranel who said it worked as well.  I wonder if there was a breaking change in the meantime, especially given how long it took for the pull request to get merged.

mklemarczyk, 2022-09-14:
If testing is required. I can perform with various devices (Arduino Nano BLE, Nano IoT, Nano Connect).
I can test with Windows 10 and Linux hosted on Raspberry Pi 3 and Pi 4.

vadimkhm, 2024-05-28:
I wonder if someone can figure out this, I have the same problem but with Arduino Uno R4 Wifi.

In my case, BLE is connected to Windows if I  call this _`BLE.setPairable(0);`_.
So it works only for Windows, for Android phones _`BLE.setPairable(1);`_ should be called.