arduino-libraries/ArduinoBLE #249: BLE is doing with hardFault exception when using CAN ISRs

fterretf, 2022-08-17:
My project is using CAN and BLE at the same time on Arduino mkr1010
BLE alone is OK
BLE and CAN without ISR is OK
BLE and CAN with CAN Rx ISRs is crashing after a while with a hardFault exception because _discoveredDevices ._root.next...next pointer is overwritten.
Any ideas?
[image]

fterretf, 2022-08-17:
More investigation: I modify the CAN interrupt to be the minimum. It now only clear the interrupt flag and wait for a while:
[image]
ModifyRegister() is clearing the interrupt flag through SPI
[image]
Is BLE using SPI? with beginTransaction() endTransaction()?

facchinm, 2022-08-23:
Hi @fterretf ,
On MKR WIFI 1010 the BLE uses the UART pins so there shouldn't be any "concurrency" issue. 
Are you using Sandeep's arduino-CAN library?
Since the crash doesn't happen immediately I'd check if there's a memory leak somewhere; some of them were solved in version 1.3.1 of this library, maybe double check that it's the version you are using :wink:

fterretf, 2022-08-23:
Hi @facchinm,
Yes, I used my own fork [fterretf](https://github.com/fterretf) /[arduino-CAN](https://github.com/fterretf/arduino-CAN) from Sandeep's arduino-CAN library.
But the problem should be reproducible with Sandeep Can lib 
Yes, I use BLE 1.3.1