arduino-libraries/ArduinoBLE #253: Reading advertising data via arduinoble

Kaly380, 2022-08-23:
Good morning, I have a Xiomi brand scale that has a ServiceData that sends the data

How can I get the data from the ServideData of the scale

Please help

[image]

per1234, 2022-08-23:
Hi @Kaly380. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/c/using-arduino/networking-protocols-and-devices/27

closed 2022-08-23 by per1234

Kaly380, 2022-08-30:
There is only the functionality implemented but that arduino peripheral mode.

Someone please guide me to work arduino central mode, and I can access the servicedata method and obtain the data of my device