arduino-libraries/ArduinoBLE #266: Receiving ACK response for BLEIndicate / BLENotify

strukiii, 2022-10-05:
Hey! For my project I have four Nano 33 BLE peripherals writing IMU data to a python bleak central on my laptop. To my understanding, BLE cannot establish truly concurrent connections to all devices.  Rather, it fakes it by rapidly swapping between the four connections. As my data updates at a high frequency (60hz for all 4 devices) this results in gaps in the data as the central rotates between peripherals. My solution is a buffer on the peripherals and to pop off the buffer once an Indicate ACK (nowledgement) is sent.  Meanwhile, the peripheral should continuously push IMU sample updates to the back of the buffer at its true refresh rate.  It is ok if the BLE transmitter lags behind and needs some time to finish the entire buffer.  

Now to my question, how do I handle the ACK response after I write a value? And subsequently handle the event with a callback? I've tried `imuCharacteristic.setEventHandler(BLERead, imuCharacteristicWritten)` though `BLERead` is not the correct event. The only remaining ones are `BLEWritten`, `BLESubscribe` and `BLEUnsubscribe` which aren't of much use.  I've also attempted to use the returned value from `BLECharacteristic.writeValue()` however it always returns 1 presumably because the return status does not indicate the value was read and acknowledged, only that it was written.

Thanks for reading, any help would be greatly appreciated!

per1234, 2022-10-05:
Hi @strukiii. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/

closed 2022-10-05 by per1234

strukiii, 2022-10-06:
My apologies, I did make a forum post here 8 days ago https://forum.arduino.cc/t/arduinoble-indicate-handle-ack-event/1036246