arduino-libraries/ArduinoBLE #273: esp32 resets to bootloader when calling BLE.end() before BLE.begin()

mvandeg-git, 2022-11-02:
Thanks for adding ESP32 support! There seems to be one quirk: When calling BLE.end() before BLE.begin(), the whole device resets and starts the bootloader. Tested on a Beetle-ESP2-C3 from DF-Robot.

On an Arduino this does not seem to cause any issues and BLE gets reset (when BLE.begin() had been called before)/or nothing noticeable happens (when BLE.begin()).

I do not know if BLE.end() also resets the device when BLE.end() is called after BLE.begin(), since I loose the USB Serial connection after BLE has been initialized.

Thanks for looking into this.

dominsch, 2022-11-17:
Can you provide a use case where you would need to call end() before begin()? If used correctly it doesn't reset the device if that is your concern.

mvandeg-git, 2022-11-18:
Hi Dominsch,

Thanks for your reply.
Perhaps I am interpreting some functionality incorrectly.

I envisioned a scenario like this:

BLE setup with BLE.begin()
Regular activity
Something goes wrong or user (soft) resets device
BLE needs to be reset: BLE.end()

I had BLE.end() in an init state function to start with a clean slate
regardless of previous initialization.
Since there is (as far as I could find) no way to determine the BLE status
after BLE.begin(), this was a bit of an ugly hack.
I've solved it now by maintaining a status variable.

I mostly filed this, because it was surprising behavior to me that the
whole device was reset (in the ESP32 case).

I hope this helps a bit.

Best,

Martijn