arduino-libraries/ArduinoBLE #274: Lots of Compiler Warnings - At least if I build for a Teensy 4.1 or Micromod

KurtE, 2022-11-16:
As I mentioned on a PJRC forum post, I have a version of this library building and running on Teensy 4.1 and Micromod using the USBHost_T36 library and a couple of edits to this library.  In particular disable the Serial transport and code for its own Transport. 
Which now appears to be working.  At least with a few of your examples. Like the ones to do with LEDs.

While doing this, I am seeing a lot of compiler warnings in the code.  And I believe most of them are not Teensy specific.  Note: I am using the latest Teensyduino IDE beta which we are trying a more recent version of GCC which is showing up a lot more compiler warning than earlier ones. 

Some of the warnings:
```
"C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\teensy\\tools\\teensy-compile\\11.3.1-beta1/arm/bin/arm-none-eabi-g++" -c -O2 -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=158 -DARDUINO=10607 -DARDUINO_TEENSY_MICROMOD -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Users\\kurte\\AppData\\Local\\Temp\\arduino-sketch-43071A1C59D1A4F494A82C8D8FA363A4/pch" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\teensy\\hardware\\avr\\0.58.2\\cores\\teensy4" "-Ic:\\Users\\kurte\\Documents\\Arduino\\libraries\\ArduinoBLE\\src" "-Ic:\\Users\\kurte\\Documents\\Arduino\\libraries\\USBHost_T36" "-Ic:\\Users\\kurte\\Documents\\Arduino\\libraries\\SDFat\\src" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\teensy\\hardware\\avr\\0.58.2\\libraries\\SPI" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\teensy\\hardware\\avr\\0.58.2\\libraries\\EEPROM" "c:\\Users\\kurte\\Documents\\Arduino\\libraries\\ArduinoBLE\\src\\remote\\BLERemoteService.cpp" -o "C:\\Users\\kurte\\AppData\\Local\\Temp\\arduino-sketch-43071A1C59D1A4F494A82C8D8FA363A4\\libraries\\ArduinoBLE\\remote\\BLERemoteService.cpp.o"
In file included from c:\Users\kurte\Documents\Arduino\libraries\ArduinoBLE\src/utility/GATT.h:26,
                 from c:\Users\kurte\Documents\Arduino\libraries\ArduinoBLE\src\local\BLELocalCharacteristic.cpp:25:
c:\Users\kurte\Documents\Arduino\libraries\ArduinoBLE\src/local/BLELocalCharacteristic.h: In constructor 'BLELocalCharacteristic::BLELocalCharacteristic(const char*, uint16_t, int, bool)':
c:\Users\kurte\Documents\Arduino\libraries\ArduinoBLE\src/local/BLELocalCharacteristic.h:90:12: warning: 'BLELocalCharacteristic::_cccdValue' will be initialized after [-Wreorder]
   90 |   uint16_t _cccdValue;
      |            ^~~~~~~~~~
c:\Users\kurte\Documents\Arduino\libraries\ArduinoBLE\src/local/BLELocalCharacteristic.h:79:12: warning:   'uint8_t BLELocalCharacteristic::_permissions' [-Wreorder]
   79 |   uint8_t  _permissions;
      |            ^~~~~~~~~~~~
c:\Users\kurte\Documents\Arduino\libraries\ArduinoBLE\src\local\BLELocalCharacteristic.cpp:32:1: warning:   when initialized here [-Wreorder]
   32 | BLELocalCharacteristic::BLELocalCharacteristic(const char* uuid, uint16_t permissions, int valueSize, bool fixedLength) :
      | ^~~~~~~~~~~~~~~~~~~~~~
"C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\teensy\\tools\\teensy-compile\\11.3.1-beta1/arm/bin/arm-none-eabi-g++" -c -O2 -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=158 -DARDUINO=10607 -DARDUINO_TEENSY_MICROMOD -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Users\\kurte\\AppData\\Local\\Temp\\arduino-sketch-43071A1C59D1A4F494A82C8D8FA363A4/pch" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\teensy\\hardware\\avr\\0.58.2\\cores\\teensy4" "-Ic:\\Users\\kurte\\Documents\\Arduino\\libraries\\ArduinoBLE\\src" "-Ic:\\Users\\kurte\\Documents\\Arduino\\libraries\\USBHost_T36" "-Ic:\\Users\\kurte\\Documents\\Arduino\\libraries\\SDFat\\src" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\teensy\\hardware\\avr\\0.58.2\\libraries\\SPI" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\teensy\\hardware\\avr\\0.58.2\\libraries\\EEPROM" "c:\\Users\\kurte\\Documents\\Arduino\\libraries\\ArduinoBLE\\src\\utility\\GATT.cpp" -o "C:\\Users\\kurte\\AppData\\Local\\Temp\\arduino-sketch-43071A1C59D1A4F494A82C8D8FA363A4\\libraries\\ArduinoBLE\\utility\\GATT.cpp.o"
```
And:
```
"C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\teensy\\tools\\teensy-compile\\11.3.1-beta1/arm/bin/arm-none-eabi-g++" -c -O2 -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=158 -DARDUINO=10607 -DARDUINO_TEENSY_MICROMOD -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Users\\kurte\\AppData\\Local\\Temp\\arduino-sketch-43071A1C59D1A4F494A82C8D8FA363A4/pch" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\teensy\\hardware\\avr\\0.58.2\\cores\\teensy4" "-Ic:\\Users\\kurte\\Documents\\Arduino\\libraries\\ArduinoBLE\\src" "-Ic:\\Users\\kurte\\Documents\\Arduino\\libraries\\USBHost_T36" "-Ic:\\Users\\kurte\\Documents\\Arduino\\libraries\\SDFat\\src" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\teensy\\hardware\\avr\\0.58.2\\libraries\\SPI" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\teensy\\hardware\\avr\\0.58.2\\libraries\\EEPROM" "c:\\Users\\kurte\\Documents\\Arduino\\libraries\\ArduinoBLE\\src\\utility\\keyDistribution.cpp" -o "C:\\Users\\kurte\\AppData\\Local\\Temp\\arduino-sketch-43071A1C59D1A4F494A82C8D8FA363A4\\libraries\\ArduinoBLE\\utility\\keyDistribution.cpp.o"
c:\Users\kurte\Documents\Arduino\libraries\ArduinoBLE\src\utility\HCI.cpp: In member function 'virtual void HCIClass::poll(long unsigned int)':
c:\Users\kurte\Documents\Arduino\libraries\ArduinoBLE\src\utility\HCI.cpp:141:20: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
  141 |     if (_recvIndex >= sizeof(_recvBuffer)) {
      |         ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
c:\Users\kurte\Documents\Arduino\libraries\ArduinoBLE\src\utility\HCI.cpp: In member function 'virtual int HCIClass::tryResolveAddress(uint8_t*, uint8_t*)':
c:\Users\kurte\Documents\Arduino\libraries\ArduinoBLE\src\utility\HCI.cpp:565:14: warning: unused variable 'memcheck' [-Wunused-variable]
  565 |     uint8_t* memcheck;
      |              ^~~~~~~~
c:\Users\kurte\Documents\Arduino\libraries\ArduinoBLE\src\utility\HCI.cpp:557:11: warning: variable 'irk' set but not used [-Wunused-but-set-variable]
  557 |   uint8_t irk[16];
      |           ^~~
c:\Users\kurte\Documents\Arduino\libraries\ArduinoBLE\src\utility\L2CAPSignaling.cpp: In member function 'virtual void L2CAPSignalingClass::handleSecurityData(uint16_t, uint8_t, uint8_t*)':
c:\Users\kurte\Documents\Arduino\libraries\ArduinoBLE\src\utility\L2CAPSignaling.cpp:271:8: warning: unused variable 'pairingFailed' [-Wunused-variable]
  271 |     } *pairingFailed = (PairingFailed*)data;
      |        ^~~~~~~~~~~~~
c:\Users\kurte\Documents\Arduino\libraries\ArduinoBLE\src\utility\L2CAPSignaling.cpp:320:7: warning: unused variable 'readPublicKeyCommand' [-Wunused-variable]
  320 |     } readPublicKeyCommand = {
      |       ^~~~~~~~~~~~~~~~~~~~
c:\Users\kurte\Documents\Arduino\libraries\ArduinoBLE\src\utility\L2CAPSignaling.cpp: In member function 'virtual void L2CAPSignalingClass::smCalculateLTKandConfirm(uint16_t, uint8_t*)':
c:\Users\kurte\Documents\Arduino\libraries\ArduinoBLE\src\utility\L2CAPSignaling.cpp:421:19: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
  421 |     for(int i=0; i<sizeof(Eb); i++){
      |                  ~^~~~~~~~~~~
c:\Users\kurte\Documents\Arduino\libraries\ArduinoBLE\src\utility\HCI.cpp: In member function 'virtual void HCIClass::handleEventPkt(uint8_t, uint8_t*)':
c:\Users\kurte\Documents\Arduino\libraries\ArduinoBLE\src\utility\HCI.cpp:979:13: warning: unused variable 'BD_ADDRs' [-Wunused-variable]
  979 |     uint8_t BD_ADDRs[num_keys][6];
      |             ^~~~~~~~
[... cut ...]
Found 20:24:93:53:d2:12 '' 
Found 40:ae:a8:59:63:5b '' 
Found 4d:82:4e:f5:6c:96 '' 
Found 22:7c:49:50:eb:21 '' fd6f
Found 78:ca:b4:94:96:d6 '' 
Found 67:12:a5:9c:57:ad '' 
Found d8:ec:5e:82:11:30 'Linksys' 00002080-8eab-46c2-b788-0e9440016fd1
Found e1:59:47:79:f2:be 'Surface Arc Mouse' 1812
Connecting ...
Connected
Discovering attributes ...
Attribute discovery failed!

Device name: 
Appearance: 0x0

Service Count: 0

Disconnecting ...
Disconnected
```
But that is probably outside of this issue.

mcl-uk, 2023-05-06:
Hi KurtE, did you ever resolve these issues? I'm seeing the same kind of stuff.

KurtE, 2023-05-08:
Sorry @mci-uk I have not looked at it for a while now.   It was sort more of a random project I was working on, although it might be nice to figure out the best way to use this on a Teensy.  But I was happy that it at least worked, where I could plug a BT dongle into the USBHost port of T3.6 or T4.x and be able to use this library and talk to PC over BLE.

mcl-uk, 2023-05-08:
I have since made good progress, I fixed most of the compile-time warnings, mostly to do with obsolete debug code (which I also removed) then I hacked HCI.cpp so that it was hard-wired to Serial2 and got rid of HCITransport.*, also sorted out wiring configuration using a new h file. And fixed the baud rate to 115200, after that it finally worked - at least enough for my application. I'll try to get round to posting my hacked version of the ardiunoBLE lib for other Teensy users when I have time.

per1234, 2024-01-28:
I'm closing this as a duplicate of https://github.com/arduino-libraries/ArduinoBLE/issues/310. Even though this one was submitted first, the other contains up to date information about the progress on fixing the code that produces warnings (fortunately most of them have already been fixed via https://github.com/arduino-libraries/ArduinoBLE/pull/302).

closed 2024-01-28 by per1234