arduino-libraries/ArduinoBLE #276: XBox One controller - current firmware (BLE) - (peripheral.discoverAttributes fails

KurtE, 2022-11-18:
If you update an XBox One controller to the latest firmware, the update will convert the controller to running using BLE.
Microsoft shows a few different ways to do the update, including I believe simply plug it into USB to XBox One that is running the latest stuff.  I updated one of mine using the Windows XBox Accessories App. 

Side note: trying to add BLE support to our Teensy USBHost_t36 code base.  We already have some bluetooth support, and for example we can connect and use an XBox one controller, with the original firmware, but not after they are updated.

So I thought I would experiment using the ArduinoBLE library, and first tried on Arduino Nano 33 IOT as well as BLE and this call fails. 
Note: I now also have a fork/branch of this library that runs on Teensy boards (T3.6 and 4.x) and right now doing most of my stuff
I am using a bluetooth dongle plugged into an USBHost connector on either a 4.1 or a Micromod.   

In order for BLE to see this you need to put the controller into pairing mode.  Hopefully after I figure out this part will then figure out how to do an actual BLE pairing (Again sort of outside of this issue).

When in pairing mode I found a windows App, that can talk to the device.
[image]

Note: I posted Some of these details up on both a PJRC forum thread as well as an Arduino one:
https://forum.arduino.cc/t/arduinoble-trying-to-connect-to-xbox-one-controller-and-other-ble-devices/1054844/3
https://forum.pjrc.com/threads/71503-USBHost-Bluetooth-gt-BLE-gt-ArduinoBLE

As the title mentioned, the discoverAttributes call fails.   I have added a lot of debug outputs and the like to code including into my
TeensyTransport code.  This includes doing some decoding of the data that is sent back and forth with the ATT messages.

It looks like the code is properly enumerating the Services and the Characteristics and then fails after that:

From my debug output: first enumerating the Services:
Pardon some of my cryptic outputs:
```
>>(ACLDATA, 0):40 00 07 00 03 00 04 00 02 12 00 
	** MTU_REQ **: MTU: 12
<<(ACLDATA, 5183):40 20 07 00 03 00 04 00 03 12 00 
	** MTU_RESP **: MTU: 12
>>(ACLDATA, 42):40 00 0B 00 07 00 04 00 10 01 00 FF FF 00 28 
	** READ_BY_GROUP_REQ **: Starting handle:0001 ending handle:ffff Group Type:2800
<<(ACLDATA, 45):40 20 12 00 0E 00 04 00 11 06 01 00 07 00 00 18 08 00 08 00 01 18 
	** READ_MULTI_RESP **: len per:06
		Atr Handle:0001 End Group Handle:0012 Data: 00 18
		Atr Handle:0008 End Group Handle:0012 Data: 01 18
>>(ACLDATA, 29):40 00 0B 00 07 00 04 00 10 09 00 FF FF 00 28 
	** READ_BY_GROUP_REQ **: Starting handle:0009 ending handle:ffff Group Type:2800
<<(ACLDATA, 30):40 20 12 00 0E 00 04 00 11 06 09 00 11 00 0A 18 12 00 15 00 0F 18 
	** READ_MULTI_RESP **: len per:06
		Atr Handle:0009 End Group Handle:0012 Data: 0a 18
		Atr Handle:0012 End Group Handle:0012 Data: 0f 18
>>(ACLDATA, 30):40 00 0B 00 07 00 04 00 10 16 00 FF FF 00 28 
	** READ_BY_GROUP_REQ **: Starting handle:0016 ending handle:ffff Group Type:2800
<<(ACLDATA, 30):40 20 0C 00 08 00 04 00 11 06 16 00 23 00 12 18 
	** READ_MULTI_RESP **: len per:06
		Atr Handle:0016 End Group Handle:000c Data: 12 18
>>(ACLDATA, 14):40 00 0B 00 07 00 04 00 10 24 00 FF FF 00 28 
	** READ_BY_GROUP_REQ **: Starting handle:0024 ending handle:ffff Group Type:2800
<<(ACLDATA, 45):40 20 09 00 05 00 04 00 01 10 24 00 80 
	** ATT_ERROR **: OP:10 handle:24 error:80 

	After discoverServices
```
But it looks here like it found:
Handle:0001 ID: 1800
0008 ID: 1801
0009  180a
0012  180f 
0016 1812

Which map to:
```
      case 0x1800: return "Generic Access service";
      case 0x1801: return "Generic Attribute service";
      case 0x180A: return "Device Information service";
      case 0x1812: return "Human Interface Device service";

```
It then goes through all 5 of these Services looking for Characteristics:
[... 126 lines cut ...]
<<(EVT, 2, H:171 T:171)0F 04 00 01 06 04 
HCI event: F
F n cmd:  0x1
F status: 0x0
F opcode: 0x406
<<(EVT, 1899, H:178 T:178)05 04 00 40 00 22 
HCI event: 5
Command tx ->  0x1 0xA 0x20 0x1 0x1
>>(CMD, 0):0A 20 01 01 
    Control callback (bluetooth): 0 : 0a 20 01 01 
<<(EVT, 2, H:185 T:185)0E 04 01 0A 20 00 
HCI event: E
E ncmd:   0x1
E opcode: 0x200A
E status: 0x0
`
```
Something that does not feel right here is:
As I mentioned earlier is it looks like the first one has handle 2
but it is searching for it with the range 4,5? 
It finds the 2nd one with ID 4, 
It then tries a search on 5-5 which then bails with an error and I don't
see any response on the ACLDATA for it.

Still investigating

KurtE, 2022-11-29:
I know I am probably simply talking to myself (not too unusual) ;)

I now have my extended version of the Peripheral Explorer getting information from an XBox One Controller with the updated firmware that uses BLE on a Teensy 4.1... 
Example Output:
```
HCIClass::readLeBufferSize called
Bluetooth® Low Energy Central - Peripheral Explorer
Found c8:3f:26:b3:ac:16 'Xbox Wireless Controller' 
Connecting ...
Connected
Discovering attributes ...
Attributes discovered

Device name: Xbox Wireless Controll
Appearance: 0x0

Service 1800	//Generic Access service
	Characteristic 2a00, properties 0x2, value 0x58626F7820576972656C65737320436F6E74726F6C6C	//Device Name
		Descriptor 2803, value 0x020300002A	//Characteristic
		Descriptor 00002a00-0000-0000-0000-000000000000, value 0x58626F7820576972656C65737320436F6E74726F6C6C
	Characteristic 2a01, properties 0x2, value 0xC403	//Appearance
		Descriptor 2803, value 0x020500012A	//Characteristic
		Descriptor 00002a01-0000-0000-0000-000000000000, value 0xC403
	Characteristic 2a04, properties 0x2, value 0x0600060000002C01	//Peripheral Preferred Connection Parameters
		Descriptor 2803, value 0x020700042A	//Characteristic
		Descriptor 00002a04-0000-0000-0000-000000000000, value 0x0600060000002C01
Service 1801	//Generic Attribute service
Service 180a	//Device Information service
	Characteristic 2a29, properties 0x2, value 0x4D6963726F736F6674	//Manufacturer Name String
		Descriptor 2803, value 0x020B00292A	//Characteristic
		Descriptor 00002a29-0000-0000-0000-000000000000, value 0x4D6963726F736F6674
	Characteristic 2a50, properties 0x2, value 0x025E04200B1505	//PnP ID
		Descriptor 2803, value 0x020D00502A	//Characteristic
		Descriptor 00002a50-0000-0000-0000-000000000000, value 0x025E04200B1505
	Characteristic 2a26, properties 0x2, value 0x352E31352E333136382E30	//Firmware Revision String
		Descriptor 2803, value 0x020F00262A	//Characteristic
		Descriptor 00002a26-0000-0000-0000-000000000000, value 0x352E31352E333136382E30
	Characteristic 2a25, properties 0x2, value 0x3033363030303332393035363234	//Serial Number String
		Descriptor 2803, value 0x021100252A	//Characteristic
		Descriptor 00002a25-0000-0000-0000-000000000000, value 0x3033363030303332393035363234
Service 180f	//Battery service
	Characteristic 2a19, properties 0x12, value 0x32	//Battery Level
		Descriptor 2803, value 0x121400192A	//Characteristic
		Descriptor 00002a19-0000-0000-0000-000000000000, value 0x32
		Descriptor 2902, value 0x0000	//Client Characteristic Configuration
Service 1812	//Human Interface Device service
	Characteristic 2a4a, properties 0x2	//HID Information
		Descriptor 2803, value 0x0218004A2A	//Characteristic
		Descriptor 00002a4a-0000-0000-0000-000000000000, value 0x
	Characteristic 2a4c, properties 0x4	//HID Control Point
		Descriptor 2803, value 0x041A004C2A	//Characteristic
		Descriptor 00002a4c-0000-0000-0000-000000000000, value 0x
	Characteristic 2a4b, properties 0x2	//Report Map
		Descriptor 2803, value 0x021C004B2A	//Characteristic
		Descriptor 00002a4b-0000-0000-0000-000000000000, value 0x
	Characteristic 2a4d, properties 0x12	//Report
		Descriptor 2803, value 0x121E004D2A	//Characteristic
		Descriptor 00002a4d-0000-0000-0000-000000000000, value 0x
		Descriptor 2902, value 0x0000	//Client Characteristic Configuration
		Descriptor 2908, value 0x0101	//Report Reference
	Characteristic 2a4d, properties 0xE	//Report
		Descriptor 2803, value 0x0E22004D2A	//Characteristic
		Descriptor 00002a4d-0000-0000-0000-000000000000, value 0x
		Descriptor 2908, value 0x0302	//Report Reference
Service 00000001-5f60-4c4f-9c83-a7953298d40d
	Characteristic 00000002-5f60-4c4f-9c83-a7953298d40d, properties 0x2
		Descriptor 2803, value 0x0226000DD4983295A7839C4F4C605F02000000	//Characteristic
		Descriptor 00000002-5f60-4c4f-9c83-a7953298d40d, value 0x
	Characteristic 00000003-5f60-4c4f-9c83-a7953298d40d, properties 0x2
		Descriptor 2803, value 0x0228000DD4983295A7839C4F4C605F03000000	//Characteristic
		Descriptor 00000003-5f60-4c4f-9c83-a7953298d40d, value 0x
	Characteristic 00000004-5f60-4c4f-9c83-a7953298d40d, properties 0x8
		Descriptor 2803, value 0x082A000DD4983295A7839C4F4C605F04000000	//Characteristic
		Descriptor 00000004-5f60-4c4f-9c83-a7953298d40d, value 0x

Disconnecting ...
Disconnected

```
My Fork/Branch is updated with the code.  Still has some debug stuff turned on.

KurtE, 2022-11-29:
Note: The main Issue that was causing this to not work, was in the function:
```
int HCIClass::readLeBufferSize(uint16_t& pktLen, uint8_t& maxPkt)
{
  int result = sendCommand(OGF_LE_CTL << 10 | OCF_LE_READ_BUFFER_SIZE);

  //Serial.println("HCIClass::readLeBufferSize called");
  if (result == 0) {
    struct __attribute__ ((packed)) HCILeBufferSize {
      uint16_t pktLen;
      uint8_t maxPkt;
    } *leBufferSize = (HCILeBufferSize*)_cmdResponse;

    pktLen = leBufferSize->pktLen;
    _maxPkt = maxPkt = leBufferSize->maxPkt;

#ifdef _BLE_TRACE_
  Serial.print("\tpktlen: ");
  Serial.print(pktLen, DEC);
  Serial.print(" maxPkt: ");
  Serial.println(maxPkt, DEC);
#endif

#if (defined(TEENSYDUINO) && (__IMXRT1062__) || (ARDUINO_TEENSY36)) || !defined(__AVR__)
    ATT.setMaxMtu(pktLen/* - 9*/); // max pkt len - ACL header size
#endif
  }

  return result;
}
```
The Bluetooth 4.0 dongle returned 27 and the code was subtracting 9 from it for the ACL data size...

From the Bluetooth 5.3 manual it says:

> Total_Num_ISO_Data_Packets return parameter contains the total number of
isochronous data packets that can be stored in the data buffers of the
Controller. The Host determines how to divide the buffers between different
connection handle(s).
Note: The LE_ACL_Data_Packet_Length and ISO_Data_Packet_Length
return parameters do not include the length of the HCI ACL Data packet header
or the HCI ISO Data packet header respectively.

Removing that -9 caused everything to appear to work now.

bediver, 2023-01-08:
Hello, I've run into same issue when attempting discoverAttributes on XBox controller (BLE). I've been testing your changes and what made this work are your changes for descriptor UUID (16/128 bit format).
-9 on setMaxMtu has no effect in my case.
My HW: Arduino Nano RB2040 Connect

Stoutscientist, 2023-02-24:
^

fstrugar, 2024-01-16:
Hi! Just run into the same issue with Arduino Uno R4 - stuck in infinite loop in discoverAttributes. Can confirm what @bediver said, your changes in discoverDescriptors from https://github.com/KurtE/ArduinoBLE/tree/uno_r4_wifi_teensy branch fix it. 

Thank you kindly for putting it up here, I would've never figured it out myself! Hope ArduinoBLE devs will pick it up eventually?