arduino-libraries/ArduinoBLE #277: peripheral.discoverAttributes() hangs when peripheral has more than one characteristics

fredsco, 2022-11-30:
**Peripheral**: Nicla sense ME
**Central**: RP2040

I wrote code for the **Nicla Sense ME** which creates one service and exactly one characteristic. This characteristic is added to the service. I then write a value to the characteristics. The **RP2040** is then able to receive and print this value. 

When I add another characteristic to the Nicla sense me, the **RP2040** hangs and the LED flickers RED when it comes to "Discovering attributes".

I attached part of the code where I add characteristics on the Peripheral and do a "discoverAttributes" on the Central.

### **CODE**:

**Peripheral**: 
 1/2

```
BLEService aService("0x01FF");
BLEUnsignedIntCharacteristic temperatureValChar("0000FFE1-0000-0000-0000-00000000FF01", BLERead | BLENotify);
BLEUnsignedIntCharacteristic humidityValChar("0000FFE2-0000-0000-0000-00000000FF01", BLERead | BLENotify);
```
 
2/2

**Note**:  here when I remove the `aService.addCharacteristic(humidityValChar);` the central won't hang during discoverAttributes and everything works fine.

```
BLE.setAdvertisedService(aService);
aService.addCharacteristic(temperatureValChar);
aService.addCharacteristic(humidityValChar);
```

**Central**: 

1/2

```
Serial.println("Connecting ...");
  if (peripheral.connect())
  {
    Serial.println("Connected");
  }
  else
  {
    Serial.println("Failed to connect!");
    return;
  }
  // discover peripheral attributes
  Serial.println("Discovering attributes ...");
  if (peripheral.discoverAttributes())
  {
    Serial.println("Attributes discovered");
  }
  else
  {
    Serial.println("Attribute discovery failed!");
    peripheral.disconnect();
    return;
  }
```

2/2

```
BLECharacteristic TempCharacteristic = peripheral.characteristic("00000000-0000-0000-0000-00000000FF01");
BLECharacteristic HumidCharacteristic = peripheral.characteristic("00000000-0000-0000-0000-00000000FF02");
```

### **OUTPUT**:

**Peripheral**:

[image]

**Central**:

[image]

paulvha, 2022-12-04:
The problem is related to the size of the UUID. A client should be able to handle BOTH 16-bit descriptor UUID and 128 BIT UUID.
The 16-bit is handled correctly, but the calculation for 128-bit is incorrect. 

In the file src/utility/ATT.CPP, in the function  ATTClass::discoverDescriptors(),  around line 1729, change

`uint16_t lengthPerDescriptor = responseBuffer[1] * 4;`

to
```
          // 3.4.3.2 Find Information Response
          uint16_t lengthPerDescriptor = 4 ; // assume 16-bit UUID ( 2 bytes + 2 handle)
          if (responseBuffer[1] == 0x2) lengthPerDescriptor = 18; // if 128 BIT UUID (16 + 2 handle)
```

Regards,
Paul

fredsco, 2022-12-12:
Thanks @paulvha once I started using 128-bit UUID there were no issues anymore!

closed 2022-12-12 by fredsco