arduino-libraries/ArduinoBLE #281: BLE communication on MKR1010, after some hours central will not discover any peripherals

AndreaScavuzzo, 2023-01-06:
hello,
I have implemented a communication between two Arduino MKR1010 using ArduinoBLE:

example from ArduinoBLE library:
PeripheralExplorer (central)
Battery monitor (peripheral)

Initially after reset all is working, **but after one ore more hours the central doesn't recognize any Bluetooth peripheral**.

here the code but they are from the library example:
[BLE_example.zip](https://github.com/arduino-libraries/ArduinoBLE/files/10362040/BLE_example.zip)

(I have tried both latest firmware WIFI NINA 1.4.8 and 1.2.4 but with the same results, seems that with the 1.2.4 the behaviour is better)

I think that this is a very strange behaviour **and is not accetable to have it from Arduino,** how I can produce a project if after some hour of running nothing is working?

thanks,
Andrea

AndreaScavuzzo, 2023-01-08:
hello,
I think that the issue is similar to the ticket: **Fix BLE scanning block after few minutes #264**

some fixes has been done on repository but they are not present in the official latest TAG (1.3.2),

With the latest files from master my issue is not yet present,
are you generating a new TAG with the file fixed?

thanks,
Andrea

per1234, 2023-01-09:
Thanks for taking the time to post an update @AndreaScavuzzo. Since the issue has already been resolved, I will close this as resolved.

As you pointed out, a production release has not been made since the time of the fix, but this issue tracker is used to track work to be done in the codebase hosted in this repository, so we close issues as soon as they are resolved in the codebase; not when the resolution is released.

closed 2023-01-09 by per1234