arduino-libraries/ArduinoBLE #285: Version problem

FilipeFaria90, 2023-02-01:
Hello, I'm having some problems with the new version (1.3.2), if I use version 1.3.1, I don't have any problem.
But when I compile with the new version I get this error log

In file included from c:\Users\fimartin\Documents\Arduino\libraries\ArduinoBLE\src\utility\HCIVirtualTransport.cpp:22:0:
c:\Users\fimartin\Documents\Arduino\libraries\ArduinoBLE\src\utility\HCIVirtualTransport.h:27:36: fatal error: freertos/stream_buffer.h: No such file or directory
compilation terminated.
Se encontraron varias bibliotecas para "BLEDevice.h"
  Usado: C:\Users\fimartin\Documents\Arduino\libraries\ArduinoBLE
  No utilizado: C:\Users\fimartin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\BLE
exit status 1

Compilation error: exit status 1

Can you tell me what I am doing wrong?
I even updated version after version to tried solve it, but I couldn't fix it this way.

per1234, 2023-02-01:
Hi @FilipeFaria90. The reason for this error is you are using an outdated version of the ESP32 boards platform. This library is not compatible with the version 1.0.6 you are using.

The solution is to use the Arduino IDE Boards Manager to update the platform. I'll provide instructions:

1. Select **File > Preferences** from the Arduino IDE menus.
   The "**Preferences**" dialog will open.
1. Click the icon on the right side of the "**Additional Boards Manager URLs**" field.
   An "**Additional Boards Manager URLs**" dialog will open. The field contains a list of any URLs you already configured.
1. If the following URL is present on the list, delete it:
   ```text
   https://dl.espressif.com/dl/package_esp32_index.json
   ```
   **ⓘ** This is the old URL which Espressif no longer maintains so it doesn't provide the new versions of the platform.
1. Add the following URL on a new line in the field:
   ```text
   https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
   ```
1. Click the **<kbd>OK</kbd>** button on the "**Additional Boards Manager URLs**" dialog.
1. Click the **<kbd>OK</kbd>** button on the "**Preferences**" dialog.
1. You will now see a "**Downloading index: ...**" notification at the bottom right corner of the IDE window. Wait for that notification to close.
1. Select **Tools > Board > Boards Manager** from the Arduino IDE menus to open the "**Boards Manager**".
1. Wait for the IDE to finish any index update operations, as shown in the notification area.
1. Scroll down through the list of boards platforms until you see the "**esp32 by Espressif Systems**" entry.
1. Make sure version **2.0.6** is selected from the version menu on the entry.
1. Click the "**INSTALL**" button.
1. Wait for the installation to finish.

You should now be able to compile sketches using the **ArduinoBLE** library for ESP32 boards without getting that "`freertos/stream_buffer.h: No such file or directory`" error.

closed 2023-02-01 by per1234

FilipeFaria90, 2023-02-02:
Hello per1234, after doing as you say, I compile the file I have this log

In file included from C:\Users\fimartin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\BLE\src/BLECharacteristic.h:17,
                 from C:\Users\fimartin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\BLE\src/BLEService.h:15,
                 from C:\Users\fimartin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\BLE\src/BLEClient.h:20,
                 from C:\Users\fimartin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\BLE\src/BLEUtils.h:16,
                 from C:\Users\fimartin\Documents\Repo_GIT\esp32_BLE\esp32_BLE.ino:15:
C:\Users\fimartin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\BLE\src/BLEDescriptor.h:25:7: error: redefinition of 'class BLEDescriptor'
 class BLEDescriptor {
       ^~~~~~~~~~~~~
In file included from c:\Users\fimartin\Documents\Arduino\libraries\ArduinoBLE\src/BLECharacteristic.h:25,
                 from c:\Users\fimartin\Documents\Arduino\libraries\ArduinoBLE\src/BLEService.h:23,
                 from c:\Users\fimartin\Documents\Arduino\libraries\ArduinoBLE\src/BLEDevice.h:25,
                 from C:\Users\fimartin\Documents\Repo_GIT\esp32_BLE\esp32_BLE.ino:14:
c:\Users\fimartin\Documents\Arduino\libraries\ArduinoBLE\src/BLEDescriptor.h:28:7: note: previous definition of 'class BLEDescriptor'
 class BLEDescriptor {
       ^~~~~~~~~~~~~
In file included from C:\Users\fimartin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\BLE\src/BLEService.h:15,
                 from C:\Users\fimartin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\BLE\src/BLEClient.h:20,
                 from C:\Users\fimartin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\BLE\src/BLEUtils.h:16,
                 from C:\Users\fimartin\Documents\Repo_GIT\esp32_BLE\esp32_BLE.ino:15:
C:\Users\fimartin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\BLE\src/BLECharacteristic.h:53:7: error: redefinition of 'class BLECharacteristic'
 class BLECharacteristic {
       ^~~~~~~~~~~~~~~~~
In file included from c:\Users\fimartin\Documents\Arduino\libraries\ArduinoBLE\src/BLEService.h:23,
                 from c:\Users\fimartin\Documents\Arduino\libraries\ArduinoBLE\src/BLEDevice.h:25,
                 from C:\Users\fimartin\Documents\Repo_GIT\esp32_BLE\esp32_BLE.ino:14:
c:\Users\fimartin\Documents\Arduino\libraries\ArduinoBLE\src/BLECharacteristic.h:45:7: note: previous definition of 'class BLECharacteristic'
 class BLECharacteristic  {
       ^~~~~~~~~~~~~~~~~
In file included from C:\Users\fimartin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\BLE\src/BLEClient.h:20,
                 from C:\Users\fimartin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\BLE\src/BLEUtils.h:16,
                 from C:\Users\fimartin\Documents\Repo_GIT\esp32_BLE\esp32_BLE.ino:15:
C:\Users\fimartin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\BLE\src/BLEService.h:49:7: error: redefinition of 'class BLEService'
 class BLEService {
       ^~~~~~~~~~
In file included from c:\Users\fimartin\Documents\Arduino\libraries\ArduinoBLE\src/BLEDevice.h:25,
                 from C:\Users\fimartin\Documents\Repo_GIT\esp32_BLE\esp32_BLE.ino:14:
c:\Users\fimartin\Documents\Arduino\libraries\ArduinoBLE\src/BLEService.h:28:7: note: previous definition of 'class BLEService'
 class BLEService {
       ^~~~~~~~~~
C:\Users\fimartin\Documents\Repo_GIT\esp32_BLE\esp32_BLE.ino: In member function 'virtual void SecurityCallback::onAuthenticationComplete(esp_ble_auth_cmpl_t)':
C:\Users\fimartin\Documents\Repo_GIT\esp32_BLE\esp32_BLE.ino:134:18: error: 'startAdvertising' is not a member of 'BLEDevice'
       BLEDevice::startAdvertising();
                  ^~~~~~~~~~~~~~~~
C:\Users\fimartin\Documents\Repo_GIT\esp32_BLE\esp32_BLE.ino: In member function 'virtual void MyCallbacks1::onWrite(BLECharacteristic*)':
C:\Users\fimartin\Documents\Repo_GIT\esp32_BLE\esp32_BLE.ino:169:46: error: 'class BLECharacteristic' has no member named 'getValue'; did you mean 'setValue'?
       std::string value = pCharacteristic_1->getValue();
                                              ^~~~~~~~
                                              setValue
C:\Users\fimartin\Documents\Repo_GIT\esp32_BLE\esp32_BLE.ino: In function 'void setup()':
C:\Users\fimartin\Documents\Repo_GIT\esp32_BLE\esp32_BLE.ino:250:14: error: 'init' is not a member of 'BLEDevice'
   BLEDevice::init(nameAndModel.c_str());
              ^~~~
C:\Users\fimartin\Documents\Repo_GIT\esp32_BLE\esp32_BLE.ino:251:14: error: 'setEncryptionLevel' is not a member of 'BLEDevice'
   BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT);
              ^~~~~~~~~~~~~~~~~~
C:\Users\fimartin\Documents\Repo_GIT\esp32_BLE\esp32_BLE.ino:252:14: error: 'setSecurityCallbacks' is not a member of 'BLEDevice'
   BLEDevice::setSecurityCallbacks(new SecurityCallback());
              ^~~~~~~~~~~~~~~~~~~~
C:\Users\fimartin\Documents\Repo_GIT\esp32_BLE\esp32_BLE.ino:254:24: error: 'createServer' is not a member of 'BLEDevice'
   pServer = BLEDevice::createServer();
                        ^~~~~~~~~~~~
C:\Users\fimartin\Documents\Repo_GIT\esp32_BLE\esp32_BLE.ino:256:33: error: 'class BLEService' has no member named 'createCharacteristic'; did you mean 'addCharacteristic'?
[... cut ...]
C:\Users\fimartin\Documents\Repo_GIT\esp32_BLE\esp32_BLE.ino:460:26: error: 'class BLECharacteristic' has no member named 'notify'
       pCharacteristic_2->notify();
                          ^~~~~~
Se encontraron varias bibliotecas para "BLEDevice.h"
  Usado: C:\Users\fimartin\Documents\Arduino\libraries\ArduinoBLE
  No utilizado: C:\Users\fimartin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\BLE
exit status 1

Compilation error: 'startAdvertising' is not a member of 'BLEDevice'

Do I have to change something in the code?