arduino-libraries/ArduinoBLE #287: Discussion. Clarification and examples about BLE Security

vovagorodok, 2023-02-12:
Hi,
Can't find Discussions in this project, that's why decided to create Issue.

I'm working on solution to upload/update firmware via BLE.
Arduino library:
https://github.com/vovagorodok/ArduinoBleOTA
And web/mobile/desktop application:
https://github.com/vovagorodok/ble_ota_app

There are 3 items that should be considered:
1. mandatory, Integrity protection. DONE by crc32 checking
2. optional, Version control possibility. DONE by additional characteristics
3. optional, Security possibility. Still open

Don't know for this moment how to provide this possibility to user of ArduinoBleOTA library.
Will be nice to consider it now, without changing library interface and protocol i future.
I think that following possibilities should be considered:
1. Authentication (Pin code) for whole connection, with and without transfer encryption
2. Authentication (Pin code) for only OTA service, with and without transfer encryption
3. Authentication (Pin code) for only OTA upload characteristics, with and without transfer encryption

Can't find simple example how to do that, and even don't know if each of these items are possible to implement in BLE or ArduinoBLE.
I have very small knowledge about BLE Security. Please help

per1234, 2023-02-13:
Hi @vovagorodok. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests specific to the project. This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/

If you do identify any areas for improvement or defects in the **ArduinoBLE**, you are welcome to submit specific issues here for each formal feature request or bug report.

Regarding authentication, you might be interested in the information that was shared in this feature request: https://github.com/arduino-libraries/ArduinoBLE/issues/36 (but note that thread is not an appropriate place to discuss your own project).

closed 2023-02-13 by per1234

vovagorodok, 2023-02-27:
Hi, sure, Issue is not correct place for this.
Maybe will be nice to create Discussions for this procject:
[image]

Created topic:
https://forum.arduino.cc/t/arduinoble-clarification-and-examples-about-ble-security/1093188
But no response for this moment. Currently I'm stuck with this issue.
I've spend couple days on analyzing ArduinoBLE and can't create solution for requesting pin when ble connecting.
For NimBLE-Arduino library enough:
```
BLEDevice::setSecurityPasskey(123456);
BLEDevice::setSecurityAuth(true, true, true); // bonding, mitm, sc 
BLEDevice::setSecurityIOCap(BLE_HS_IO_DISPLAY_ONLY);
```
And start security on connected callback:
```
void ArduinoBleOTAClass::onConnect(BLEServer* pServer, ble_gap_conn_desc* desc)
{
  BLEDevice::startSecurity(desc->conn_handle);
}
```

Initially at ArduinoBLE, as I see, to achieve `BLE_HS_IO_DISPLAY_ONLY` we shoud `BLE.setDisplayCode()` and not `BLE.setBinaryConfirmPairing()`.
How to start security when ble connecting?

I have plan to add additional example to ArduinoBLE for security when ble connecting. Created example unblock me to finish work at BLE firmware uploads library