arduino-libraries/ArduinoBLE #290: Unexpected behaviour of device argument in connect/disconnect callback functions

s-a-i-m-o-n, 2023-03-11:
Using the BLE library we can set callback which are called when a device connects or disconnects
`// assign event handlers`
`BLE.setEventHandler(BLEConnected, connectHandler);`
`BLE.setEventHandler(BLEDisconnected, disconnectHandler);`
The callback functions have the following form:
`void connectHandler(BLEDevice peripheral) {
Serial.println(peripheral.localName()); // prints nothing
Serial.println(peripheral.address()); // prints adress
}`
However, when accessing the ble device variable passed to the function is not working as expected. 
For example accessing .address() will return the address of the device as expected.
However, accessing .localName() will return an empty string even though when the device has a local name set.
(I can print the localName just find if I first use BLE.scanForName() and then .localName() in the setup or loop function).

Note: This is my first post on GIT-Hub, go easy on me.

rmlearney-digicatapult, 2023-04-13:
Your interrupt handler probably takes too long to execute. Instead you should just set a flag in the handler which your main loop picks up and then executes the very slow retrieval of peripheral name, address, and then serial writes.

s-a-i-m-o-n, 2023-06-02:
Sorry for my late reply. I did not find time up to know to try out your suggestion.

I trimmed the code inside the handlers and do the main work inside the main loop. I use a queue instead of flags to react to connect/disconnect/receive events inside the main loop. Everything seems to work fine now. Thank you.

closed 2023-06-02 by s-a-i-m-o-n