arduino-libraries/ArduinoBLE #293: Again Incorrect change applied for ARDUINO_GIGA

paulvha, 2023-03-27:
As mentioned in  https://github.com/arduino-libraries/ArduinoBLE/issues/292 there was a incorrect change for the ARDUINO_GIGA.
I see you have taken quick action.. but that is also incorrect. It will not work for the Apollo3 and other processors on MBED.

The  correct change for line 238:

`#if !defined(ARDUINO_PORTENTA_H7_M4) && !defined(ARDUINO_PORTENTA_H7_M7) && !defined(ARDUINO_NICLA_VISION) && !defined(ARDUINO_GIGA)`

kind regards,
Paul

facchinm, 2023-03-27:
Mmmh not sure about that; Apollo3 is not an STM32H7 board, so `terminate()` will be called (as it was before the erroneous first patch) .
Am I missing something?

paulvha, 2023-03-27:
In 1.3.2 the 238 line was :

`#if !defined(ARDUINO_PORTENTA_H7_M4) && !defined(ARDUINO_PORTENTA_H7_M7) && !defined(ARDUINO_NICLA_VISION) `

So ONLY in the case PORTENTA's or NICLA_VISION the termination was not called. With the latest change it is also called for them.  

But maybe the previous check was incorrect. !!

Did a quick check on PORTENTA and the NICLA-VSION libraries, both are using MBED as well. Too bad many of the files are not in the downloaded cores/ MBED board library to check. 

I assume you will know that, and then you are correct. 

regards,
Paul

closed 2023-03-27 by facchinm