arduino-libraries/ArduinoBLE #295: Impossible to use secure/encrypted pairing - arduino nano 33 BLE

maximouth, 2023-03-28:
Hello,

I want to use the different levels of authentication for a central to connect to a peripheral.
I've seen in the EncryptedBatteryMonitor exemple you provide l38 and 39 the way to declare a characteristic of a service which require encryption to write on it : [https://github.com/arduino-libraries/ArduinoBLE/blob/master/examples/Peripheral/EncryptedBatteryMonitor/EncryptedBatteryMonitor.ino](url)

`<// Add BLEEncryption tag to require pairing. This controls the LED`
`BLEUnsignedCharCharacteristic secretValue("2a3F", BLERead | BLEWrite | BLEEncryption);`

I've also seen in the source code that it is possible to have other option for it. [https://github.com/arduino-libraries/ArduinoBLE/blob/d5bc5b850636be9da59a6c6e147e9d2585433ee4/src/BLEProperty.h](url)

`enum BLEProperty {`
  `BLEBroadcast            = 0x01,`
  `BLERead                 = 0x02,`
  `BLEWriteWithoutResponse = 0x04,`
  `BLEWrite                = 0x08,`
  `BLENotify               = 0x10,`
  `BLEIndicate             = 0x20,`
  `BLEAuthSignedWrite      = 1 << 6,`
 ` BLEExtProp              = 1 << 7,`
`};`

`enum BLEPermission {`
 ` BLEEncryption         = 1 << 9,`
  `BLEAuthentication     = 1 << 10,`
  `BLEAuthorization      = 1 << 11,`
  `// BLEWriteEncryption        = 1 << 11,`
  `// BLEWriteAuthentication    = 1 << 12,`
  `// BLEWriteAuthorization     = 1 << 13,`
`};`

There is also the `enum BLE_GATT_PERM_ {` *(which I don't understand the usage)*

--------------

So to go back to the example, I don't understand how the callback function for the pairing works.
There are functions declared between line 58 and 151 named callbacks, and key management.
I don't understand how to use them, and where to declare them.

In the function `setEventHandler`, the only handler available are the connect and disconnect.

When I try to use this example, every time I try to pair a device to the example running, I got an error, and no evidence of the callback function called, nothing prints on the Serial port.

Can you help me with this please ? To understand how to use the different callback pairing functions, and the different parameters available and the purpose of the different existing enum and all the things I wonder into this field.
Is there documentation about how to use this feature ?
Is there another existing library usable to use secure pairing between two elements ?

Thank you in advance.
Maxime

IDC44, 2023-04-09:
Hello, 
Have you managed to use pairing with pass key?
Thanks

maximouth, 2023-04-12:
Hello,

No I gave up using the arduino library for that.. like other peope before me with the no responses we get.
I've moved to the nrf library.
As it's a NRF52840 used into the nano 33 BLE board it's good.
https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/zephyr/boards/arm/arduino_nano_33_ble/doc/index.html

It's a little more complicated to use than the arduino library, but at least it's working.
I can pair with a pass key two BLE boards/phone/object...

But if there is an existing solution and a documentation about how to use the arduino library for secure BLE communication I'm up too :)

secretshardul, 2023-09-13:
Ping, any updates?