arduino-libraries/ArduinoBLE #299: Get info about MTU and package (or overhead) size

vovagorodok, 2023-05-13:
Can't find how to determinate MTU and max package size (MTU - some overhead). Is here some method to check the MTU and overhead size?

vovagorodok, 2023-05-16:
As I see ATT has variable overhead size: https://github.com/chrisc11/ble-guides/blob/master/ble-throughput.md

vovagorodok, 2023-05-16:
1. Found  `ATTClass::mtu(handle)` where I can get handle?

paulvha, 2023-05-18:
some time ago I have this already implemented for myself. 

4 changes.

in **src/local/BLElocalDevice.cpp** add

```
uint16_t BLELocalDevice::readMTU()
{

  BLEDevice central = ATT.central();

  if (central) {
    return central.readMTU();
  }

  return 0;
}
```

in src/local/BLELocalDevice.h add in public
`virtual uint16_t readMTU();`

in **src/BLEDevice.cpp** add : 

```
uint16_t BLEDevice::readMTU()
{
  // get connectionhandle
  uint16_t handle = ATT.connectionHandle(_addressType, _address);

  if (handle == 0xffff){
    Serial.println("Error during obtaining connection handle\n");
    return 0;
  }

  // minus 1 as the first byte in a transfer unit is the opcode
  // minus 2 as a handle can be added
  //Thus minus 3
  return(ATT.mtu(handle)-3);
}
```

in **src/BLEDevice.h** add in Public 
`uint16_t readMTU(); `

From the sketch you can now call after central has been connected 

`central.readMTU()`

It always starts with an MTU 23, but can be increased later on. For a brief explanation of how MTU is negotiated see https://github.com/arduino-libraries/ArduinoBLE/issues/297

Regards,
Paul

vovagorodok, 2023-07-24:
How to check max MTU size for nano_33_iot and set it?

paulvha, 2023-07-24:
Follow the steps above to read it

Setting the max common MTU size is automatically exchanged. it is initiated by the central as described in https://github.com/arduino-libraries/ArduinoBLE/issues/297

vovagorodok, 2023-08-02:
Additional helpful info:
BLE packet types:
https://microchipdeveloper.com/wireless:ble-link-layer-packet-types
MTU overhead:
https://docs.silabs.com/bluetooth/4.0/general/system-and-performance/throughput-with-bluetooth-low-energy-technology#attribute-protocol-att-operation

Additional answers for my questions:
>  Is here some method to check the MTU size?

shown by your example
> How handle MTU negotiation procedure? Who should initialize it (central or peripheral)? Or it's done automatically during connection procedure?

Central can negotiate it. Sometimes  it made automatically during connection (for example BlueZ)

Now this issue describe all that aspects. Thanks, lets close it!

closed 2023-08-02 by vovagorodok

pschatzmann, 2023-11-13:
Could you please reopen this issue ?
From my point of view it can only be closed when the described functions have been added to the library!