arduino-libraries/ArduinoBLE #308: retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown when trying to connect to Wahoo Tickr

esranonff, 2023-07-16:
_Originally logged as https://github.com/nkolban/esp32-snippets/issues/1169_

Hey folks,

The following code worked perfectly well when the library was separate from the Arduino base libraries, however it now throws the above error.

The device is an ESP32Dev-C1, and I'm programming via platformio.

### Platformio Ini

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
; change microcontroller
board_build.mcu = esp32
build_flags = !./bin/build_flags.sh
board_build.partitions = min_spiffs.csv
lib_deps = 
	knolleary/PubSubClient@^2.8.0
;	nkolban/ESP32 BLE Arduino@^1.0.1
	bblanchon/ArduinoJson@^6.18.0

```

### buildflags.sh
```
#!/bin/bash
source .env
echo -n "
-DWIFI_PASS=${WIFI_PASS}
-DWIFI_SSID=${WIFI_SSID}
-DMQTT_SERVER=${MQTT_SERVER}
-DMQTT_USER=${MQTT_USER}
-DMQTT_PASS=${MQTT_PASS}
-DMQTT_PORT=${MQTT_PORT}
-DMQTT_PROTO=${MQTT_PROTO}
"
```

### Application Code:

```
#include "BLEDevice.h"
#include "ArduinoJson.h"
#include <WiFi.h>
#include <PubSubClient.h>
#define ST(A) #A
#define STR(A) ST(A)

char device_type[] = "heartrate_monitor";

// TypeDef
typedef struct {
  char ID[20];
  uint16_t HRM;
}HRM;
HRM hrm;
// BLE
[... 248 lines cut ...]
        Serial.println(pub_topic_name);
        // Once connected, publish an announcement...
        mqtt.publish(pub_topic_name, bodyString);
        lastReconnectAttempt = 0;
      }
    }
  } else {
    // Client connected
    mqtt.loop();
  }
}

```
### The error in full (MAC address redacted)
```
BLE Advertised Device found: Name: TICKR B444, Address: <MAC>, appearance: 833, serviceUUID: 0000180d-0000-1000-8000-00805f9b34fb
Found our device!  address: Forming a connection to <MAC>
 - Created client
 - Connected to server
 - Found our service
[  6911][E][BLERemoteCharacteristic.cpp:287] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
 - Found our characteristic

assert failed: heap_caps_free heap_caps.c:339 (heap != NULL && "free() target pointer is outside heap areas")
```

per1234, 2023-07-16:
Hi @esranonff. Thanks for taking the time to submit an issue.

You are not using the library that is hosted in this repository. Instead, you are using the library that is bundled with the esp32 boards platform installation:

https://github.com/espressif/arduino-esp32/tree/master/libraries/BLE

So I will close this issue as off topic for this repository.

If you plan to submit the issue to the esp32 boards platform repository, I urge you to first distill your code down to a [minimal, reproducible, verifiable example](https://stackoverflow.com/help/minimal-reproducible-example). Remove all the MQTT code if possible.

If you are only hoping to get assistance with your project, then you should post on Arduino Forum instead:

https://forum.arduino.cc/

closed 2023-07-16 by per1234

esranonff, 2023-07-17:
@per1234 thanks for the feedback, it gets a bit confusing when libraries have similar names but are from different places!

I'll hunt around for an alternative option.