arduino-libraries/ArduinoBLE #310: Warnings when compiling this code

roleroz, 2023-07-20:
When you build the code in this library there are several warnings printed

- Order member initialization on BLELocalCharacteristic
- Unused variables in:
  - HCIClass::tryResolveAddress
  - HCIClass::handleEventPkt (the whole else if statement did nothing)
  - L2CAPSignalingClass::handleSecurityData (one initialization should be inside of BLE_TRACE
  - BluetoothCryptoToolbox::f5
  - BluetoothCryptoToolbox::testAh
  - BluetoothCryptoToolbox::testg2
- Uninitialized variable in L2CAPSignalingClass::handleSecurityData
- Unenumerated case in ArduinoBLE/src/utility/HCI.cpp

I have a PR ready for fixing most of these warnings (all except the last one) in https://github.com/arduino-libraries/ArduinoBLE/pull/302, but I haven't had luck getting any traction on getting that reviewed

If someone could review that PR most of these things would be fixed. Also, if someone tells me what 0x0A means in that unenumerated case, I'm happy to add that to the enum as well

roleroz, 2023-08-04:
This is still incomplete, there is also the unenumerated case in ArduinoBLE/src/utility/HCI.cpp, if someone can tell me what does 0x0A means in that file, I'll create a PR for it

ataweg, 2024-01-17:
Issue "Unenumerated case in ArduinoBLE/src/utility/HCI.cpp" **not fixed** . Depending on the Preferences "Compiler warnings" set to All or More there is an error message: 
```
f:\Projects\Arduino\libraries\ArduinoBLE\src\utility\HCI.cpp: In member function 'virtual void HCIClass::handleEventPkt(uint8_t, uint8_t*)':
f:\Projects\Arduino\libraries\ArduinoBLE\src\utility\HCI.cpp:991:7: error: case value '10' not in enumerated type 'LE_META_EVENT' [-Werror=switch]
       case 0x0A:{
       ^~~~
cc1plus.exe: some warnings being treated as errors
```

fpistm, 2025-03-14:
This 0x0A value was introduced with PR #156 
It seems to be a the same than `CONN_COMPLETE` with only 2 fields added in the struct:
```
          uint8_t localResolvablePrivateAddress[6];
          uint8_t peerResolvablePrivateAddress[6];
```
But never filled. Only print when _BLE_TRACE_ is enabled.

Maybe @unknownconstant could comment on this?

unknownconstant, 2025-03-21:
Let me take a look - it's been a while since I've been through this code

fpistm, 2025-03-21:
I think I've found which LE META EVENT it is.
It seems to be the LE Enhanced Connection Complete Event.

[image]

I would propose this as a fix:
```patch
----------------------------- src/utility/HCI.cpp -----------------------------
index 0e64063..2394fbb 100644
@@ -993,7 +993,7 @@ void HCIClass::handleEventPkt(uint8_t /*plen*/, uint8_t pdata[])
     Serial.println(leMetaHeader->subevent,HEX);
 #endif
     switch((LE_META_EVENT)leMetaHeader->subevent){
-      case 0x0A:{
+      case ENHANCED_CONN_COMPLETE:{
         struct __attribute__ ((packed)) EvtLeConnectionComplete {
           uint8_t status;
           uint16_t handle;

------------------------------ src/utility/HCI.h ------------------------------
index 0a530ce..a6fa66e 100644
@@ -46,7 +46,8 @@ enum LE_META_EVENT {
   LONG_TERM_KEY_REQUEST     = 0x05,
   REMOTE_CONN_PARAM_REQ     = 0x06,
   READ_LOCAL_P256_COMPLETE  = 0x08,
-  GENERATE_DH_KEY_COMPLETE  = 0x09
+  GENERATE_DH_KEY_COMPLETE  = 0x09,
+  ENHANCED_CONN_COMPLETE    = 0x0A,
 };
 String metaEventToString(LE_META_EVENT event);
 String commandToString(LE_COMMAND command);
```

closed 2025-04-08 by facchinm