arduino-libraries/ArduinoBLE #311: Arduino R4 WIFI: If a Servo is added to a BLE sketch, incorrect UUID is returned by peripheral

MrYsLab, 2023-07-20:
I documented how I tested BLE Uart [here](https://forum.arduino.cc/t/ble-uart-works/1148164).  An incorrect Service UUID is returned if the sketch is modified to add a servo. The modified sketch is shown below. Also, a Python script is provided to act as a BLE central and to print out what is being returned from the Arduino R4.

I don't know if this is related to issue #309.

The expected UUID should be UART_SERVICE_UUID = "6E400001-B5A3-F393-E0A9-E50E24DCCA9E"

```
#include <HardwareBLESerial.h>

// Uncomment these next two lines to demonstrate bug
#include <Servo.h>
Servo myServo ;

HardwareBLESerial &bleSerial = HardwareBLESerial::getInstance();

void setup() {
  Serial.begin(115200);
  while (!Serial);

  if (!bleSerial.beginAndSetupBLE("SerialPassthrough")) {
    while (true) {
      Serial.println("failed to initialize HardwareBLESerial!");
      delay(1000);
    }
  }

  // wait for a central device to connect
  while (!bleSerial);

  Serial.println("HardwareBLESerial central device connected!");
}

void loop() {
  // this must be called regularly to perform BLE updates
  bleSerial.poll();

  // whatever is written to BLE UART appears in the Serial Monitor
  while (bleSerial.available() > 0) {
    Serial.write(bleSerial.read());
  }

  // whatever is written in Serial Monitor appears in BLE UART
  while (Serial.available() > 0) {
    bleSerial.write(Serial.read());
  }
}

```

```

"""
 Copyright (c) 2020-2023 Alan Yorinks All rights reserved.

 This program is free software; you can redistribute it and/or
 modify it under the terms of the GNU AFFERO GENERAL PUBLIC LICENSE
 Version 3 as published by the Free Software Foundation; either
 or (at your option) any later version.
 This library is distributed in the hope that it will be useful,
 but WITHOUT ANY WARRANTY; without even the implied warranty of
 MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
 General Public License for more details.

 You should have received a copy of the GNU AFFERO GENERAL PUBLIC LICENSE
 along with this library; if not, write to the Free Software
 Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
"""

import asyncio
import sys
import bleak

from bleak import BleakClient, BleakScanner
from bleak.backends.characteristic import BleakGATTCharacteristic
from bleak.backends.device import BLEDevice
from bleak.backends.scanner import AdvertisementData

UART_SERVICE_UUID = "6E400001-B5A3-F393-E0A9-E50E24DCCA9E"
UART_RX_CHAR_UUID = "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
UART_TX_CHAR_UUID = "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"

class TelemetrixAioBle:
    """
    This class encapsulates management of a BLE UART connection that communicates
    with an Arduino UNO R4 WIFI
    """
    def __init__(self, ble_device_name="SerialPassthrough"):
        self.ble_device_name = ble_device_name
        self.ble_device = None
        self.bleak_client = None

    async def connect(self):
        """
        This method connects to a device matching the ble_device_name

        :return:
        """
        print(f'Scanning for BLE device {self.ble_device_name}.  Please wait...')

        self.ble_device = await BleakScanner.find_device_by_name(self.ble_device_name)
        if self.ble_device is None:
            raise RuntimeError('Did not find the BLE device. Please check name.')
        print(f'Found  {self.ble_device_name}  address: {self.ble_device.address}')
        self.bleak_client = BleakClient(self.ble_device.address)
        await self.bleak_client.connect()
        print('Connected')
        print(self.ble_device.details)

        sys.exit(0)

theBle = TelemetrixAioBle()

asyncio.run(theBle.connect())

```

Here is the output of the Python script with the servo added. The item to look at is UUIDs. It is clearly not the UART_SERVICE_UUID.

```
Scanning for BLE device SerialPassthrough.  Please wait...
Found  SerialPassthrough  address: DC:54:75:C3:BD:CD
Connected
{'path': '/org/bluez/hci0/dev_DC_54_75_C3_BD_CD', 'props': {'Address': 'DC:54:75:C3:BD:CD', 'AddressType': 'public', 'Name': 'SerialPassthrough', 'Alias': 'SerialPassthrough', 'Paired': False, 'Bonded': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb'], 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False, 'RSSI': -70}}
```

If I comment out the lines adding the servo, here is the output of the Python script:

```
Scanning for BLE device SerialPassthrough.  Please wait...
Found  SerialPassthrough  address: DC:54:75:C3:BD:CD
Connected
{'path': '/org/bluez/hci0/dev_DC_54_75_C3_BD_CD', 'props': {'Address': 'DC:54:75:C3:BD:CD', 'AddressType': 'public', 'Name': 'SerialPassthrough', 'Alias': 'SerialPassthrough', 'Paired': False, 'Bonded': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -73, 'Connected': False, 'UUIDs': ['6e400001-b5a3-f393-e0a9-e50e24dcca9e'], 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False}}
```

Here the UUID is correct.

```
#include <HardwareBLESerial.h>

// Uncomment these next two lines to demonstrate bug
#include <Servo.h>
Servo myServo ;

HardwareBLESerial &bleSerial = HardwareBLESerial::getInstance();

void setup() {
  Serial.begin(115200);
  while (!Serial);

  if (!bleSerial.beginAndSetupBLE("SerialPassthrough")) {
    while (true) {
      Serial.println("failed to initialize HardwareBLESerial!");
      delay(1000);
    }
  }

  // wait for a central device to connect
  while (!bleSerial);

  Serial.println("HardwareBLESerial central device connected!");
}

void loop() {
  // this must be called regularly to perform BLE updates
  bleSerial.poll();

  // whatever is written to BLE UART appears in the Serial Monitor
  while (bleSerial.available() > 0) {
    Serial.write(bleSerial.read());
  }

  // whatever is written in Serial Monitor appears in BLE UART
  while (Serial.available() > 0) {
    bleSerial.write(Serial.read());
  }
}

```

facchinm, 2023-09-16:
Should be fixed by https://github.com/arduino/ArduinoCore-renesas/commit/96963d8574956e6fc02e56666ecdbc215dbfdf84 . @MrYsLab can you retest with latest library + core?

MrYsLab, 2023-09-16:
@facchinm I tested, and yes, it is fixed. Thank you. I am closing this issue.

closed 2023-09-16 by MrYsLab