arduino-libraries/ArduinoBLE #322: Version 1.3.6 still doesn't work with UNO R4 WiFi in real life

ArduinoManager, 2023-08-31:
Environment:

Arduino IDE: 2.2.0
ArduinoBLE: 1.3.6

This is my test code
`
#include <ArduinoBLE.h>
#include <BLECharacteristic.h>

BLEService mainService = BLEService("19B10000-E8F2-537E-4F6C-D104768A1214");
BLECharacteristic rxCharacteristic = BLECharacteristic("19B10001-E8F2-537E-4F6C-D104768A1214", BLEWrite, 20, true);
BLECharacteristic txCharacteristic = BLECharacteristic("19B10002-E8F2-537E-4F6C-D104768A1214", BLERead | BLENotify, 20, true);

BLEService batteryService = BLEService("180F");
BLEUnsignedCharCharacteristic batteryLevelChar = BLEUnsignedCharCharacteristic("2A19", BLERead | BLENotify);

void setup() {
  Serial.begin(115200);
  while (!Serial)
    
  delay(500);
  Serial.println("Initializing BLE module");

  if (!BLE.begin()) {
    Serial.println("starting BLE failed!");
    return;
  }
  BLE.setDeviceName("Bug Test");
  BLE.setLocalName("Bug Test");
  BLE.setAdvertisedService(mainService);

  mainService.addCharacteristic(rxCharacteristic);
  delay(1000);

  mainService.addCharacteristic(txCharacteristic);
  delay(1000);

  BLE.addService(mainService);
  delay(1000);

  batteryService.addCharacteristic(batteryLevelChar);
  delay(1000);

  BLE.addService(batteryService);
  delay(1000);

  BLE.setEventHandler(BLEConnected, connectHandler);
  BLE.setEventHandler(BLEDisconnected, disconnectHandler);
  rxCharacteristic.setEventHandler(BLEWritten, characteristicWritten);

  BLE.advertise();

  Serial.println("Advertising started");
}

void loop() {

  BLE.poll();
  delay(200);
}

void connectHandler(BLEDevice central) {
  Serial.print("Connected event, central: ");
  Serial.println(central.address());
}

void disconnectHandler(BLEDevice central) {
  Serial.print("Disconnected event, central: ");
  Serial.println(central.address());
}

void characteristicWritten(BLEDevice central, BLECharacteristic characteristic) {
  Serial.println("Characteristic event, written: ");

  char buffer[40];
  int n = characteristic.readValue(buffer, sizeof(buffer));
  memset(&buffer[n], 0, sizeof(buffer) - n);
  String d = String(buffer);

  Serial.print("R >");
  Serial.print(d);
  Serial.print("< ");
  Serial.println(n);
}
`
**Arduino Nano 33 IoT**: Code correctly publish services and characteristics.
**Arduino R4 WiFi: Only the Battery Level services shows up on LightBlue for iOS

Reported here as well: [https://forum.arduino.cc/t/arduinoble-doesnt-work-inside-library/1148979](url). 

I simplified the test code removing the library but it still doesn't work.

facchinm, 2023-08-31:
The library needs an updated version of the Renesas core (namely 1.0.3) which solved the heap corruption bug https://github.com/arduino/ArduinoCore-renesas/commit/96963d8574956e6fc02e56666ecdbc215dbfdf84 . The core will be available VERY soon :slightly_smiling_face:

friyiajr, 2023-09-15:
@facchinm - In my case `BLE.begin()` doesn't even work on Arduino UNO R4 WiFi. Is this the same issue? My code looks like this:

```
#include <ArduinoBLE.h>

const char* deviceServiceUuid = "19b10000-e8f2-537e-4f6c-d104768a1214";
const char* deviceServiceCharacteristicUuid = "19b10001-e8f2-537e-4f6c-d104768a1214";

BLEService gestureService(deviceServiceUuid); 
BLEByteCharacteristic gestureCharacteristic(deviceServiceCharacteristicUuid, BLERead | BLEWrite);

void setup() {
  Serial.begin(9600);
  while (!Serial);  
  
  if (!BLE.begin()) {
    Serial.println("- Starting Bluetooth® Low Energy module failed!");
    while (1);
  }

  BLE.setLocalName("Friyia - Arduino R4 WiFi BLE (Peripheral)");
  BLE.setAdvertisedService(gestureService);
  gestureService.addCharacteristic(gestureCharacteristic);
  BLE.addService(gestureService);
  gestureCharacteristic.writeValue(-1);
  BLE.advertise();

  Serial.println("Arduino R4 WiFi BLE (Peripheral Device)");
  Serial.println(" ");
}

void loop() {
  BLEDevice central = BLE.central();
  Serial.println("- Discovering central device...");
  delay(500);

  if (central) {
    Serial.println("* Connected to central device!");
    Serial.print("* Device MAC address: ");
    Serial.println(central.address());
    Serial.println(" ");

    while (central.connected()) {
    }
    
    Serial.println("* Disconnected to central device!");
  }
}
```

And it prints: `- Starting Bluetooth® Low Energy module failed!`

facchinm, 2023-09-17:
Fixed by core 1.0.3 release

closed 2023-09-17 by facchinm