arduino-libraries/ArduinoBLE #325: Issue with a new board

VEGA-Processor, 2023-10-05:
Hello everyone,
I am using the new development board called ARIES IoT. This board is built upon a RISC-V ISA compliant VEGA Processor. I did the necessary changes in BLELocalDevice::begin() function to initialise all the pins. I am trying peripheral/LED example to control a built-in LED with nRF Connect app. 
It is working when I put some print messages before HCITransport.write() function call in HCIClass::sendCommand() function. I tried to replace the print statements with delay but it is not working that way. 
```
int HCIClass::sendCommand(uint16_t opcode, uint8_t plen, void* parameters)
{
  struct __attribute__ ((packed)) {
    uint8_t pktType;
    uint16_t opcode;
    uint8_t plen;
  } pktHdr = {HCI_COMMAND_PKT, opcode, plen};

  uint8_t txBuffer[sizeof(pktHdr) + plen];
  memcpy(txBuffer, &pktHdr, sizeof(pktHdr));
  memcpy(&txBuffer[sizeof(pktHdr)], parameters, plen);

  if (_debug) {
    dumpPkt("HCI COMMAND TX -> ", sizeof(pktHdr) + plen, txBuffer);
  }
#ifdef _BLE_TRACE_
  Serial.print("Command tx -> ");
  for(int i=0; i< sizeof(pktHdr) + plen;i++){
    Serial.print(" 0x");
    Serial.print(txBuffer[i],HEX);
    
  }
  Serial.println("");
#endif
  Serial.println(__LINE__);
  Serial.println(__LINE__);
  Serial.println(__LINE__);
  Serial.println(__LINE__);
  // delayMicroseconds(1389);
  // Serial.println();
  delay(1);

  HCITransport.write(txBuffer, sizeof(pktHdr) + plen);

  _cmdCompleteOpcode = 0xffff;
  _cmdCompleteStatus = -1;

  for (unsigned long start = millis(); _cmdCompleteOpcode != opcode && millis() < (start + 1000);) {
    poll();
  }

  return _cmdCompleteStatus;
}
```
I am using the latest ArduinoBLE library. 

I added following lines in begin():
```
  pinMode(SPIWIFI_SS, OUTPUT);
  pinMode(NINA_RESETN, OUTPUT);
  
  digitalWrite(SPIWIFI_SS, LOW);

  digitalWrite(NINA_RESETN, LOW);
  delay(100);
  digitalWrite(NINA_RESETN, HIGH);
  delay(750);

// set SS HIGH
  digitalWrite(SPIWIFI_SS, HIGH);

  // set RTS HIGH
  pinMode(NINA_RTS, OUTPUT);
  digitalWrite(NINA_RTS, HIGH);

  // set CTS as input
  pinMode(NINA_CTS, INPUT);
```

Baudrate is 115200

per1234, 2023-10-05:
Hi @VEGA-Processor. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests specific to the project. This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/

closed 2023-10-05 by per1234

VEGA-Processor, 2023-10-05:
OK, thank you @per1234