arduino-libraries/ArduinoBLE #328: Get notification from HID characteristic

SimonFilgis, 2023-10-15:
Dear all,

I connect to a smart remote with 5 buttons. Pheripheral explorer shows the following HID service characteristics:

Service 1812
       Characteristic 2a4e, properties 0x6, value 0x01
               Descriptor 2803, value 0x1A27004D2A
               Descriptor 2a4d, value 0x
       Characteristic 2a4d, properties 0x1A
               Descriptor 2902, value 0x0000
               Descriptor 2908, value 0x0101
               Descriptor 2803, value 0x1A2B004D2A
               Descriptor 2a4d, value 0x
       Characteristic 2a4d, properties 0x1A
               Descriptor 2902, value 0x0000
               Descriptor 2908, value 0x0201
               Descriptor 2803, value 0x0E2F004D2A
               Descriptor 2a4d, value 0x
       Characteristic 2a4d, properties 0xE
               Descriptor 2908, value 0x0102
               Descriptor 2803, value 0x1A32004D2A
               Descriptor 2a4d, value 0x
       Characteristic 2a4d, properties 0x1A
               Descriptor 2902, value 0x0000
               Descriptor 2908, value 0x0301
               Descriptor 2803, value 0x1A36004D2A
               Descriptor 2a4d, value 0x
       Characteristic 2a4d, properties 0x1A
               Descriptor 2902, value 0x0000
               Descriptor 2908, value 0x0401
               Descriptor 2803, value 0x1A3A004D2A
               Descriptor 2a4d, value 0x
       Characteristic 2a4d, properties 0x1A
               Descriptor 2902, value 0x0000
               Descriptor 2908, value 0x0501
               Descriptor 2803, value 0x1A3E004D2A
               Descriptor 2a4d, value 0x
       Characteristic 2a4d, properties 0x1A
               Descriptor 2902, value 0x0000
               Descriptor 2908, value 0x0601
               Descriptor 2803, value 0x0242004B2A
               Descriptor 2a4b, value 0x050C0901A1018503150025017501950809CD09B509B609EA09E9097209320A23028106C0

Except index 2 all of them are subscribable, but, after sucessfull subscription valueUpdated() never returns true.

Is there anything I oversee?

Best,

Simon

per1234, 2023-10-16:
Hi @per1234. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests specific to the project. This topic is more appropriate for the Arduino Forum. I see you already created a topic over there:

https://forum.arduino.cc/t/ble-get-notification-from-hid-characteristic/1178719

closed 2023-10-16 by per1234

gmacario, 2023-10-16:
> Hi @per1234. Thanks for your interest in this open source project.

Maybe you meant @SimonFilgis 😉

SimonFilgis, 2023-10-16:
@per1234  and @gmacario Thanks for your messages.

I fully understand that github issue tracker is not for discussion.
Can you please help me on the arduino forum to get my application up and running?

https://forum.arduino.cc/t/ble-get-notification-from-hid-characteristic/1178719