arduino-libraries/ArduinoBLE #332: Incorect baud rate for Nano Connect RP2040?

paulhamsh, 2023-10-27:
Hi
Although it *works* on the Nano Connect RP2040, this code has this in HCIUartTransport.cpp 
(https://github.com/arduino-libraries/ArduinoBLE/blob/master/src/utility/HCIUartTransport.cpp)

```
#if defined(ARDUINO_AVR_UNO_WIFI_REV2) || defined(ARDUINO_NANO_RP2040_CONNECT)
HCIUartTransportClass HCIUartTransport(SerialHCI, 119600);
#else
HCIUartTransportClass HCIUartTransport(SerialHCI, 912600);
#endif
```

And the NINA firmware has this in sketch.ino.cpp:
(https://github.com/arduino/nina-fw/blob/master/main/sketch.ino.cpp)

```
#if defined(UNO_WIFI_REV2) || defined(NANO_RP2040_CONNECT)
  btControllerConfig.hci_uart_baudrate = 115200;
#else
  btControllerConfig.hci_uart_baudrate = 912600;
#endif
```

So should it be changed to 115200?

facchinm, 2023-10-30:
The baud was set to 119k due to the drift of the internal crystal of the Nina module (which made the RX from the module borderline when the serial port host side was set at 115200). On most of the platforms, using any of this values doesn't show any difference since the core tries to find the best clocks divisors to approximate the requested baud, but on Uno WiFi Rev2 it was a necessary fix.

per1234, 2023-10-30:
We'll leave this open to track the need for someone to add a comment documenting this important information to avoid this otherwise unintuitive code causing the same confusion to others in the future.