arduino-libraries/ArduinoBLE #333: Immediate Panic on Portenta H7 M4 Core

trylaarsdam, 2023-10-28:
I'm trying to use BLE on the Portenta H7 M4 core (which I believe should be possible since the M4 has access to the HCI UART lines, but if someone knows definitively that it is not possible please let me know), however when I include the library, and flash to the Portenta, when I call `BLE.begin();` from the M4 core, it immediately causes an MbedOS panic.

Here's a sample that is reproducible on my end:
```cpp
// M4 Core
#include "Arduino.h"
#include "ArduinoBLE.h"

void setup() {
     BLE.begin(); // panics
}
void loop() {

}
```

There are mentions to the M4 core in the library and no mentions in Arduino's docs that the M4 core is not able to perform BLE operations, so any pointers or suggestions on how to initialize the BLE radio from the M4 core would be appreciated.

facchinm, 2023-10-30:
Fix is WIP here https://github.com/arduino-libraries/ArduinoBLE/pull/331

trylaarsdam, 2023-10-30:
Fantastic, good to know.

closed 2024-02-16 by aentinger

trylaarsdam, 2024-02-19:
@aentinger Just tested this again since it was merged and this issue is not fixed on the Portenta. The board doesn't immediately panic now, but it just freezes on `HCI.begin()` which is called by `BLE.begin()`

Here is my source code which should provide an easy test case to reproduce. This is with the src folder of the current master branch of this repo in my src folder.

```cpp
#ifdef CORE_CM4
#include <Arduino.h>
// includes from BLE lib 
#include "utility/ATT.h"
#include "utility/HCI.h"
#include "utility/GAP.h"
#include "utility/GATT.h"
#include "utility/L2CAPSignaling.h" 
#include "local/BLELocalDevice.h"
#define BT_REG_ON PJ_12
void setup()
{
	// BLE.begin();
	pinMode(LEDB, OUTPUT);
	pinMode(LED_BUILTIN, OUTPUT);
	digitalWrite(LED_BUILTIN, HIGH);
	digitalWrite(LEDB, HIGH);

	// from BLE.begin()
	pinMode(BT_REG_ON, OUTPUT);
	digitalWrite(BT_REG_ON, LOW);
	delay(500);
	digitalWrite(BT_REG_ON, HIGH);
	delay(500);

	if (!HCI.begin())
	{
		// never reached
		digitalWrite(LED_BUILTIN, LOW);
	}

	// BLE.begin continues here, but it doesn't get here anyway...

	// never reached
	digitalWrite(LEDB, LOW);
}

void loop()
{
}

#endif

#ifdef CORE_CM7
#include <Arduino.h>

void setup()
{
	bootM4();
}

void loop()
{
}
#endif
```

facchinm, 2024-02-21:
@trylaarsdam can you try https://github.com/arduino-libraries/ArduinoBLE/pull/353 on top of current master branch? In the M4 refactor we did indeed miss some defines and the module was never turned on on Portenta

trylaarsdam, 2024-02-21:
@facchinm unfortunately no change in behavior when I use your branch. When on the M4 core BLE.begin still freezes indefinitely when HCI.begin is called.

PR #353 (2024-03-05): Portenta_H7_M4: provide the correct define
Fixes #333 again (not only for Giga)
commit 99159f05 (2024-03-05): Portenta_H7_M4: provide the correct define

Fixes #333 again (not only for Giga)