arduino-libraries/ArduinoBLE #335: setValue() notifies twice

vovagorodok, 2023-12-10:
In https://github.com/vovagorodok/ArduinoBleOTA/blob/main/src/ArduinoBleOtaClass.cpp:
```
BLECharacteristic txCharacteristic(BLE_OTA_CHARACTERISTIC_UUID_TX, BLERead | BLENotify, BLE_OTA_MAX_ATTR_SIZE);
...
txCharacteristic.setValue(data, length);
```
notifies twice about written value.
Checked on esp32dev and nano_33_iot

vovagorodok, 2024-02-03:
Can't find the reason in ArduinoBLE code. Any idea why it happens? In the same scenario with NimBLE-Arduino lib I have one notification. With this issue I can't implement ArduinoBleOTA lib correctly

osemenyuk-114, 2024-02-19:
Hi! I faced the same issue. After some investigations, I suppressed echoing data back to the client. 
To implement this change you need to add BLEWriteWithoutResponse to the BLE Characteristics permissions and update code for void BLELocalCharacteristic::writeValue(BLEDevice device, const uint8_t value[], int length) function in src/local/BLELocalCharacteristic.cpp file.
`
void BLELocalCharacteristic::writeValue(BLEDevice device, const uint8_t value[], int length)
{
  _written = true;

  if (_properties & BLEWriteWithoutResponse) {
    _valueLength = min(length, _valueSize);
    memcpy(_value, value, _valueLength);

    if (_fixedLength) {
      _valueLength = _valueSize;
    }
  } 
  else {
    writeValue(value, length);
  }

  if (_eventHandlers[BLEWritten]) {
    _eventHandlers[BLEWritten](device, BLECharacteristic(this));
  }
}
`