arduino-libraries/ArduinoBLE #361: Bluetooth error

0cococ, 2024-06-10:
I used BLEDevice.h to turn on Bluetooth. My development board is esp32s3. Then I connected this Bluetooth with an Android 10 machine. At the moment of connection, Arduino reported an error. How can I solve it?
[image]

E (23013) BT_SMP: smp_calculate_link_key_from_long_term_key failed to update link_key. Sec Mode = 2, sm4 = 0x00
E (23013) BT_SMP: smp_derive_link_key_from_long_term_key failed

E (23019) BT_BTM: btm_proc_smp_cback received for unknown device
E (23236) BT_BTM: BTM_GetSecurityFlags false

E (23326) BT_BTM: BTM_GetSecurityFlags false

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

// 为BLE服务和特征定义UUID
#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

void setup() {
  Serial.begin(115200);

  // 初始化BLE并设置设备名称
  BLEDevice::init("ESP32_BLE_Server");

  // 创建BLE服务器
  BLEServer *pServer = BLEDevice::createServer();

  // 创建BLE服务
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // 创建BLE特征
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                        CHARACTERISTIC_UUID,
                        BLECharacteristic::PROPERTY_READ |
                        BLECharacteristic::PROPERTY_WRITE
                      );

  // 启动服务
  pService->start();

  // 启动广告
  pServer->getAdvertising()->start();
  Serial.println("BLE服务器已准备就绪！");
}

void loop() {
  // 服务器代码在这里
}

0cococ, 2024-06-10:
[image]

per1234, 2024-06-10:
Hi @0cococ. I'm going to ask you to provide some additional information that might help us to identify the problem.

---

❗ This procedure is not intended to solve the problem. The purpose is to gather more information.

---

Please do this:

1. Select **File > Preferences...** from the Arduino IDE menus.
   The "**Preferences**" dialog will open.
1. Check the box next to "**Show verbose output during: > ☐ compile**" in the "**Preferences**" dialog.
1. Click the "**OK**" button.
   The "**Preferences**" dialog will close.
1. Select **Sketch > Verify/Compile** from the Arduino IDE menus.
1. Wait for the compilation to finish.
1. Right click on the black "**Output**" panel at the bottom of the Arduino IDE window.
   A context menu will open.
1. Select "**Copy All**" from the context menu.
1. [Paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the error messages in a reply here.
   In order to ensure the text is not corrupted, please use [code fencing](https://guides.github.com/features/mastering-markdown/).

0cococ, 2024-06-11:
> 嗨。我将请您提供一些可能有助于我们识别问题的其他信息。
> 
> ❗ 此过程并非旨在解决问题。目的是收集更多信息。
> 
> 请这样做：
> 
> 1. 从 Arduino IDE 菜单中选择 **File > Preferences...**。将打开“**首选项**”对话框。
> 2. 选中“**首选项**”对话框中“**在：>☐编译期间显示详细输出**”旁边的框。
> 3. 单击“**确定**”按钮。“**首选项**”对话框将关闭。
> 4. 从 Arduino IDE 菜单中选择 **Sketch > Verify/Compile**。
> 5. 等待编译完成。
> 6. 右键单击Arduino IDE窗口底部的黑色“**输出**”面板。将打开一个上下文菜单。
> 7. 从上下文菜单中选择“**全部复制**”。
> 8. 将错误消息[粘贴](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525)到此处的回复中。为了确保文本不损坏，请使用[代码围栏](https://guides.github.com/features/mastering-markdown/)。

FQBN: esp32:esp32:esp32s3:CDCOnBoot=cdc
Using board 'esp32s3' from platform in folder: C:\Users\35600\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0
Using core 'esp32' from platform in folder: C:\Users\35600\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0

cmd /c if exist "C:\\Users\\35600\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2024511-4232-9d2uet.wshxk\\sketch_jun11a\\partitions.csv" COPY /y "C:\\Users\\35600\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2024511-4232-9d2uet.wshxk\\sketch_jun11a\\partitions.csv" "C:\\Users\\35600\\AppData\\Local\\Temp\\arduino\\sketches\\7E440512CC92237F2250752B66E1FBC9\\partitions.csv"
cmd /c if not exist "C:\\Users\\35600\\AppData\\Local\\Temp\\arduino\\sketches\\7E440512CC92237F2250752B66E1FBC9\\partitions.csv" if exist "C:\\Users\\35600\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0\\variants\\esp32s3\\partitions.csv" COPY "C:\\Users\\35600\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0\\variants\\esp32s3\\partitions.csv" "C:\\Users\\35600\\AppData\\Local\\Temp\\arduino\\sketches\\7E440512CC92237F2250752B66E1FBC9\\partitions.csv"
cmd /c if not exist "C:\\Users\\35600\\AppData\\Local\\Temp\\arduino\\sketches\\7E440512CC92237F2250752B66E1FBC9\\partitions.csv" COPY "C:\\Users\\35600\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0\\tools\\partitions\\default.csv" "C:\\Users\\35600\\AppData\\Local\\Temp\\arduino\\sketches\\7E440512CC92237F2250752B66E1FBC9\\partitions.csv"
cmd /c IF EXIST "C:\\Users\\35600\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2024511-4232-9d2uet.wshxk\\sketch_jun11a\\bootloader.bin" ( COPY /y "C:\\Users\\35600\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2024511-4232-9d2uet.wshxk\\sketch_jun11a\\bootloader.bin" "C:\\Users\\35600\\AppData\\Local\\Temp\\arduino\\sketches\\7E440512CC92237F2250752B66E1FBC9\\sketch_jun11a.ino.bootloader.bin" ) ELSE ( IF EXIST "C:\\Users\\35600\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0\\variants\\esp32s3\\bootloader.bin" ( COPY "C:\\Users\\35600\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0\\variants\\esp32s3\\bootloader.bin" "C:\\Users\\35600\\AppData\\Local\\Temp\\arduino\\sketches\\7E440512CC92237F2250752B66E1FBC9\\sketch_jun11a.ino.bootloader.bin" ) ELSE ( "C:\\Users\\35600\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esptool_py\\4.6\\esptool.exe" --chip esp32s3 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o "C:\\Users\\35600\\AppData\\Local\\Temp\\arduino\\sketches\\7E440512CC92237F2250752B66E1FBC9\\sketch_jun11a.ino.bootloader.bin" "C:\\Users\\35600\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083/esp32s3\\bin\\bootloader_qio_80m.elf" ) )
esptool.py v4.6
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
cmd /c if exist "C:\\Users\\35600\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2024511-4232-9d2uet.wshxk\\sketch_jun11a\\build_opt.h" COPY /y "C:\\Users\\35600\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2024511-4232-9d2uet.wshxk\\sketch_jun11a\\build_opt.h" "C:\\Users\\35600\\AppData\\Local\\Temp\\arduino\\sketches\\7E440512CC92237F2250752B66E1FBC9\\build_opt.h"
cmd /c if not exist "C:\\Users\\35600\\AppData\\Local\\Temp\\arduino\\sketches\\7E440512CC92237F2250752B66E1FBC9\\build_opt.h" type nul > "C:\\Users\\35600\\AppData\\Local\\Temp\\arduino\\sketches\\7E440512CC92237F2250752B66E1FBC9\\build_opt.h"
cmd /c type nul > "C:\\Users\\35600\\AppData\\Local\\Temp\\arduino\\sketches\\7E440512CC92237F2250752B66E1FBC9/file_opts"
Detecting libraries used...
C:\Users\35600\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-12.2.0_20230208/bin/xtensa-esp32s3-elf-g++ -c @C:\Users\35600\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32s3/flags/cpp_flags -w -Os -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32S3_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32S3_DEV" -DARDUINO_VARIANT="esp32s3" -DARDUINO_PARTITION_default -DARDUINO_HOST_OS="windows" -DARDUINO_FQBN="esp32:esp32:esp32s3:UploadSpeed=921600,USBMode=hwcdc,CDCOnBoot=cdc,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,CPUFreq=240,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=none,PSRAM=disabled,LoopCore=1,EventsCore=1,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=1 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 @C:\Users\35600\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32s3/flags/defines -IC:\Users\35600\AppData\Local\Temp\.arduinoIDE-unsaved2024511-4232-9d2uet.wshxk\sketch_jun11a -iprefix C:\Users\35600\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32s [... cut ...]
Using library BLE at version 2.0.0 in folder: C:\Users\35600\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\BLE 
"C:\\Users\\35600\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32s3-elf-gcc\\esp-12.2.0_20230208/bin/xtensa-esp32s3-elf-size" -A "C:\\Users\\35600\\AppData\\Local\\Temp\\arduino\\sketches\\7E440512CC92237F2250752B66E1FBC9/sketch_jun11a.ino.elf"
Sketch uses 927377 bytes (70%) of program storage space. Maximum is 1310720 bytes.
Global variables use 40612 bytes (12%) of dynamic memory, leaving 287068 bytes for local variables. Maximum is 327680 bytes.

per1234, 2024-06-11:
> ```text
> Using library BLE at version 2.0.0 in folder: C:\Users\35600\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\BLE 
> ```

From this part of the output you shared, we can see that you are not using the "**ArduinoBLE**" library that is hosted in this repository. You are instead using the "**BLE**" library that is bundled with the "**esp32**" boards platform:

https://github.com/espressif/arduino-esp32/tree/3.0.0/libraries/BLE

So I will close this as off topic for this issue tracker. My recommendation is for you to make a post on Arduino Forum. The Arduino community might be able to help you with the problem over there:

https://forum.arduino.cc/c/using-arduino/networking-protocols-and-devices/27

closed 2024-06-11 by per1234