arduino-libraries/ArduinoBLE #369: (support) Cannot write to BLE characteristic with vector

MaxMyzer, 2024-10-01:
System: ESP32 C3
ArduinoBLE 1.3.7

I have a system where I would like to write and read from another device. I am currently using `nRF Connect` to test functionality. 

When I check if I `canWrite`, it always returns `false` - if I do anyway, it only seems to work for a `0` value. 

I have created a demo file to demonstrate my issue: 
[ble_issue.txt](https://github.com/user-attachments/files/17198348/ble_issue.txt)

I am not quite sure what it is I am doing wrong, any help would be appreciated.

per1234, 2024-10-05:
Hi @MaxMyzer. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests specific to the project. This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/

closed 2024-10-05 by per1234

MaxMyzer, 2024-10-05:
> This issue tracker is only to be used to report bugs or feature requests specific to the project. 

Hi @per1234 
Based on the documentation, I think this actually may be a bug, since `canWrite` does not behave as expected. If it is not, I will go to the forum.